tsduck/tsduck #318: Request: combine plugin

lars18th, 2019-06-13:
Hi @lelegard ,

First of all, I know about the current `merge` and `inject` plugins. But I request for a very **simple** new plugin (I feel it's _very_ simple) for another type of merging/muxing/adding sources.

I think it's best to present it with an example:
```
tsp -P combine file1.ts file2.ts -P pcradjust -O file remux.ts
```

What's doing this `combine` plugin? It only takes 1 packet from file1 then 1 packet from file2, etc.
That's simple! For N input files it takes 1 packet of each, and if one file ends, it uses NULL stuffing to grant a constant bitrate until the last filter ends.

What's the use case?
- We want to remux (combine) two (or more) TS sources in a _deterministic_ way, without "injecting" a TS into another. This action will maintain the internal structure of the sources.
- So, we use other filters to first adapt the source files to **identical** CBR values. For example `add-input-stuffing` global parameter to increase the bitrate; `reduce` plugin to decrease the bitrate; and `limit` to cut-off the bitrate.
- In some cases, before the previous filtering we can remove some services, change tables, remap pids, etc.
- And after the "combination" we can do more postprocessing to resolve some conflits and recompute the bitrate.

In fact it's a simple "round-robin dumb remux" but called _combine_.
What you think?

lars18th, 2019-06-13:
Hi,

I forgot to mention an optional parameter: `--with-labels`. This parameter will set label 1 for packets from input 1, label 2 for input 2, and so on. Then it will be more easy to manipulate the packets after the muxing without preprocessing the inputs.

Thank you @lelegard for listening our crazy ideas! :wink:

lars18th, 2019-06-13:
Hi @lelegard ,

After thinking more on it, I see that:

1. It's an INPUT plugin, and not a processing one.
2. Two other parameters have sense: `--stop-at-first-end` and `--stop-at-last-end`. I suggest to use the last as the default. They are self explanatory (when using the second, NULL stuff is added when some input ends).
3. One _non_ trivial addition is to support not only files but other input types. However, I feel we don't need this at time.

I hope you will like to implement it.
Regards.

lelegard, 2019-06-13:
Yet another "crazy" idea indeed. I really wonder why anyone would need this.

A few remarks:

- I would prefer the name `interleave` which is more accurate (and more TVish) than `combine`.
- The best way would probably be to add some `--interleave` option to the input plugin `file`. It already accepts several files in the argument list and plays them sequentially. With this option, they would simply be interleaved.
- For other inputs than files, a `--interleave` option could be added to `tsswitch`. This command already supports several scenarios to switch between inputs. That would be just one more.

lars18th, 2019-06-14:
> Yet another "crazy" idea indeed. I really wonder why anyone would need this.

Well... I need it! The objective is a "deterministic" muxing. And I request it because I feel it's not hard to implement.

> I would prefer the name interleave which is more accurate (and more TVish) than combine.

No objection.

> The best way would probably be to add some --interleave option to the input plugin file. It already accepts several files in the argument list and plays them sequentially. With this option, they would simply be interleaved.

Great! That's just what I need. However, please consider to support the `--interlave-with-labels` option too.

> For other inputs than files, a --interleave option could be added to tsswitch. This command already supports several scenarios to switch between inputs. That would be just one more.

OK. Certainly this could be implemented in a second round.

Thank you, @lelegard, for listening to us without prejudgment. :+1:

lelegard, 2019-06-27:
Added options `--interleave`, `--label-base` and `--first-terminate` to input plugin `file`.

lars18th, 2019-07-01:
Hi @lelegard ,

Thank you for last commit with the `--interleave` functionality in the `file` Input. It works like a charm! :+1: 
So, I close this issue.

However, please consider to comment #321. Because if we mux with this "interleave mode" then we need to merge the tables. Now, I'm using something like:
```
tsp -P -I file --interleave=1 --label-base 1 --first-terminate file1.ts file2.ts file3.ts \
-P tsrename --only-label 1 -t 1 -o 1 \
-P tsrename --only-label 2 -t 1 -o 1 \
-P tsrename --only-label 3 -t 1 -o 1 \
...
```
But to merge the tables, at time, I need to do it *manually*. Please, comment about the most simple solution.
Thank you!

lars18th, 2019-07-01:
Hi @lelegard ,

I need to reopen because this doesn't work:
`tsp -a 1/2 -I file --interleave=1 --label-base 1 --first-terminate file1.ts file2.ts ...`

The idea is that for every 2 packets readed from the files, insert 1 (so if the files have the same CBR, then in fact is 1 packet file1, 1 packet file2, and 1 NULL packet). But with this command the processing is stalled at the initialization.

Please, can you check it?
Thank you!

closed 2019-07-01 by lars18th