tsduck/tsduck #319: Feature request: RTP plugin

felipewd, 2019-06-13:
Hi,

First a *great* project. We've been testing it and works like a charm for our setup.

The one thing that'd great is that our mux requires a RTP connection, and doesn't accept plain UDP.

So to use tsduck, we have `-P IP 127.0.0.1:1234` and use: `multicat -p 68 -u 127.0.0.1:1234 239.0.0.2:1010` in order to work with our mux.

I'd be great if tsduck could have a (rather simple) RTP plugin, because we can control not only the RTP configuration, but also take advantage of the `-p` options (which currently we have to control using multicat). We've tried to use FIFOs, and they work fine on the `tsp` side, but when interrupted, multicat throws an error and exit, rendering the solution unstable in case of a drop in the signal.

Kudos for this project again :-)

lelegard, 2019-06-13:
Thanks,

Having an RTP output plugin is something I had planned to do from the early days of TSDuck but had no time to do and then - unfortunately - forgot.

Maybe some day...

lelegard, 2019-06-13:
It would help if you post a real-world capture of an RTP stream transporting a TS, using Wireshark for instance. Reading an RFC is fine but matching it with the reality is better. All IPTV streams around me use plain UDP encapsulation. All IP set-top boxes I can see just use the PCR's from the input SPTS to synchronise, as they would do from a DVB stream. They do not expect any RTP header.

felipewd, 2019-06-13:
Sure can.

Attached are 2 files (200 packets each):

- in.pcap: This is the original input we're streaming - [in.pcap.gz](https://github.com/tsduck/tsduck/files/3288169/in.pcap.gz); 
- out.pcap: This is the result of the `tsp` of the in address + the `multicat` as described above [out.pcap.gz](https://github.com/tsduck/tsduck/files/3288174/out.pcap.gz).

Both are RTP streams by our mux.

Like I said, even the simplest one produced by multicat is accepted...so nothing that fancy required :-)

lelegard, 2019-06-14:
Hi,

Thanks for the files. A few remarks/questions:

- In in.pcap file, there are two UDP/RTP streams. The first one on port 1020 contains a 2 seconds TS for service CANAL 27 - VALINHOS. The TS looks fine, the service is in the clear but VLC do not show any video. The audio is ok.
- The second stream in in.pcap is on port 1022. Its content is not TS packets. Its RTP payload type is 96 (dynamic). It is not RTCP either. Any idea?
- The output stream contains the service CANAL 27 - VALINHOS but all UDP checksums are incorrect. A multicat issue? All checksum values are 0x190B.

felipewd, 2019-06-14:
Hi!

> * In in.pcap file, there are two UDP/RTP streams. The first one on port 1020 contains a 2 seconds TS for service CANAL 27 - VALINHOS. The TS looks fine, the service is in the clear but VLC do not show any video. The audio is ok.

Right, since this is only 200 packets, I suppose there could not be enough packets for a video stream. I just sent a few so you could see the RTP signature.

> * The second stream in in.pcap is on port 1022. Its content is not TS packets. Its RTP payload type is 96 (dynamic). It is not RTCP either. Any idea?

This RTP dynamic message from the original encoder to the mux (it can be ignored...such as the out.pcap doens't contain it and the mux gets the stream just fine).

> * The output stream contains the service CANAL 27 - VALINHOS but all UDP checksums are incorrect. A multicat issue? All checksum values are 0x190B.

Not, not really...this is expected since we have various offloading enabled on the NIC. I can re-capture packets with offloading disabled, (so that tcpdump will get the proper packets and check the checksum) if you think it'd be more useful.  (shouldn't affect the RTP stuff, though).

premultiply, 2019-07-21:
RTP-Header for MPEG-TS (without FEC) is very simple and almost static.
Only the sequence number field und the timestamp field are dynamic. The sync source id is a random value per session. Sequence number is a simple packet counter with overflow and and the timestamp is the systems wallclock time (90 kHz steps) at the moment the packet is written out.
[rtp.zip](https://github.com/tsduck/tsduck/files/3414377/rtp.zip)

premultiply, 2019-07-21:
For decoding the most simple way is to just drop the RTP header and handle it like UDP.
More advanced implementations use a buffer for dejittering and reordering of the received packets according to the RTP header values.
SMPTE 2022-7 advances this to send and receive multiple RTP stream paths at once and recombine them to one for redundancy passing only the the first received packet by packet counter value. This might also lower the packet jitter in special configurations.

lelegard, 2019-07-31:
RTP output has been added in commit 32b2c11455e1f3628da352acc14e6d4f35572ca0, the 24th of June. Use option `--rtp` in output plugin `ip`.

In input plugin `ip`, RTP has always been "supported", or at least ignored since all header data inside UDP packets, before the first TS packet, have always been ignored.

premultiply, 2019-07-31:
Oh, very nice! Thank you!

There is just the issue that wall clock time (should be precisely synced by NTP for best performance) as time source for RTP headers is mandatory to make dejittering, multi-path delay compensation and decoding reliable.
It is even a requirement for precise path quality measurements and monitoring.
So it would be nice if this can be improved.
Why should using the wall clock time introduce jitter?

lelegard, 2019-07-31:
> There is just the issue that wall clock time (should be precisely synced by NTP for best performance) as time source for RTP headers is mandatory to make dejittering, multi-path delay compensation and decoding reliable.

I do not understand the NTP issue. The RTP time stamps are unrelated to the wall clock time. For TS data, they just need to increment at the right rate, 90 kHz, the PTS/DTS rate. Or did I miss something?

Concerning the accuracy, TSDuck is not a production tool. It is a test, debug, demo toolbox. TSP manipulates generic transport streams coming from anywhere, with any structures and anomalies. The time source which is used by the `ip` output plugin to compute RTP time stamps is the PCR's in the stream. This is done on purpose because of the buffering and multi-thread architecture of `tsp`. The wall clock time of the system, when sending a packet, is not necessarily the right timestamp for the stream. The right clock reference is the PCR. See the file `tsIPOutputPlugin.cpp` for more details.

I assume that you are thinking in production terms. In that case, an IPTV streamer behaves the same way as a modulator. It has a pure real-time behaviour. It waits for the exact right time to send a packet and recomputes PCR's. It also computes RTP time stamps, which are consequently synchronized with the PCR's. 

In short, a production IPTV streamer generates a clock reference and derives PCR's and RTP time stamps from that source. The TSDuck `ip` output plugin, on the other hand, uses the PCR's as reference clock and derives the RTP time stamps from that source.

Production and test tools have different purposes and architectures. Production tools are specialized and work on "valid" streams. Test tools such as TSDuck are generic, flexible, address an infinity of use cases and try to adapt to any kind of more or less broken streams.

If you look for a production-grade, stable, accurate and highly synchronized RTP streamer, TSDuck is not the right choice.

premultiply, 2019-07-31:
PCRs are always a reference to a single Program clock. So it is mainly a reference to original video master clock which does not need to be in full sync to NTP (may be free running).
But in case of MPEG-TS(!) the RTP timestamp should be preferably derived from the RTP-encapsulators wall clock time.

By the way: TSDuck does a good job.

lelegard, 2019-07-31:
> PCRs are always a reference to a single Program clock. So it is mainly a reference to original video master clock 

Sorry for being pedantic for a second. In fact, it is the opposite. The PCR is the clock of the transport stream and video PTS and DTS use the PCR as reference. For instance, if you swap two TS packets, one of them containing a PCR, you need to slightly adjust the PCR value because the packet has moved along the stream. But the PTS and DTS shall not be modified.

> But in case of MPEG-TS(!) the RTP timestamp should be preferably derived from the RTP-encapsulators wall clock time.

Where is this specified? RTP and all its applications are defined in many different RFC's, so I may have missed something.

premultiply, 2019-07-31:
In case of a origin Video Encoder the PCR is directly sampled from the 27 MHz system clock which is derrived from the (SDI) input clock. PTS and DTS are used to keep multiple components (Video and (multiple) Audio) in sync and to allow for buffer management inside encoder and decoder.

> > But in case of MPEG-TS(!) the RTP timestamp should be preferably derived from the RTP-encapsulators wall clock time.
> 
> Where is this specified? RTP and all its applications are defined in many different RFC's, so I may have missed something.

It is not CLEARLY defined but it can be derrived from several publications.

Best match is still RFC 2250
https://tools.ietf.org/html/rfc2250#section-2

2. Encapsulation of MPEG System and **Transport Streams**

   Each RTP packet will contain a timestamp derived from the **sender**'s
   90KHz clock reference**.  This clock is synchronized to the system
   stream Program Clock Reference (PCR) or System Clock Reference (SCR)
_[the above is misleading as this is true for grand master clock synced video sources/encoders AND SPTS only!]_
   and **represents the target transmission time** of the first byte of the
   packet payload.  The RTP timestamp will not be passed to the MPEG
   decoder.  **This use of the timestamp is somewhat different than
   normally is the case in RTP**, in that it is **not** considered to be the
   media display or presentation timestamp. **The primary purposes of the
   RTP timestamp will be to estimate and reduce any network-induced
   jitter and to synchronize relative time drift between the transmitter
   and receiver.**

   For MPEG2 Transport Streams the RTP payload will contain an integral
   number of MPEG transport packets.  To avoid end system
   inefficiencies, data from multiple small MTS packets (normally fixed
   in size at 188 bytes) are aggregated into a single RTP packet.  The
   number of transport packets contained is computed by dividing RTP
   payload length by the length of an MTS packet (188).

More related reading matter https://www.ietf.org/proceedings/48/I-D/avt-rtp-mp2t-00.txt

lelegard, 2019-07-31:
Thanks for the pointers.

Without surprise, the IETF document focuses on networking latency issues while the ETSI document uses the PCR as possible, although not mandatory, reference.

However, the avt-rtp-mp2t-00.txt document is more specific. About CBR, it reads:

>By knowing the size of RTP packet (from the length field in UDP packet), the timestamps from the current and the next RTP packet, this constant  bitrate can be derived.

So, this means that the RTP time stamps must be synchronized with the PCR and not with the wall clock time at the sender.

premultiply, 2019-07-31:
> > By knowing the size of RTP packet (from the length field in UDP packet), the timestamps from the current and the next RTP packet, this constant  bitrate can be derived.
> 
> So, this means that the RTP time stamps must be synchronized with the PCR and not with the wall clock time at the sender.

No it does not as bitrate is defined as "bits per SECOND". Where a second is a measurable SI unit which can not be extracted "self contained" from stream. Therefore you need a least at NTP synced clock on both RTP encapsulator and decapsulator to get a drift-free measurement result of RTP transmission performance. RTP is just an additional transport layer which is independent of the MPEG-TS inside. This is what is meant by "This use of the timestamp is somewhat different than
normally is the case in RTP, in that it is not considered to be the
media display or presentation timestamp."

lelegard, 2019-07-31:
I am sorry but I do not agree.

Let's take an example: Consider a CBR stream at 9.47 Mb/s and 7 TS packets per UDP packet. This bitrate is selected because the PCR distance between two UDP packet is 100 in 90 kHz units (actual PCR distance = 100 * 300).

So:
- The PCR (in 90 kHz units) for each UDP packet is P0, P0+100, P0+200, P0+300, etc.
- The wall clock time (also in 90 kHz units) of the sending of these packets can be T0, T0+99, T0+202, T0+298, etc.

Remember that `avt-rtp-mp2t-00.txt` reads as:

> By knowing the size of RTP packet (from the length field in UDP packet), the timestamps from the current and the next RTP packet, this constant bitrate can be derived.

If the RTP time stamps contain the wall clock times (T0, T0+99, T0+202, T0+298), it does not give an accurate CBR value, unless you compute the average of a large number of samples. On the other hand, if the RTP time stamps are based on the PCR (P0, P0+100, P0+200, P0+300), then you get the correct CBR.

So, my conclusion is that you must derive the RTP time stamps from the PCR to be compliant with `avt-rtp-mp2t-00.txt`.

Now, I agree that the PCR is different from the _"media display or presentation timestamp"_ (PTS). So, the RTP time stamp is not the media display or presentation time stamp, but not the wall clock time either.

premultiply, 2019-07-31:
Just think of a MPTS from multiple free running video sources with PCR on each video stream.
They may all claim that each of them will produce 25.00000000 fps derrived from their internal (unlocked) 27 MHz reference generator. But they all drift away from each other and to the UTC wall clock time. So video source 1 may run a bit faster than source 2. But 
a mux can easily combime these signals and create a perfect new CBR MPTS without syncing to all PCRs at the same time. The Mux will spit out a new CBR TS with its own clock and his own understanding what for example 100,0 MBit/s CBR means. It may add or drop NULL packets until we thinks it is exactly the requested output bitrate.
And it will even add RTP headers to the outgoing stream by stamping each of them with his internal time reference („wall clock time“) when they are buffered to the outgoing network queue.

lelegard, 2019-07-31:
In an MPTS, each service may have its own PCR base. But all PCR streams must be synchronized on the bitrate of the global MPTS. The output plugin `ip` selects one PCR PID and only one to generate the RTP time stamps.

Generally speaking, a PCR is not based on the wall clock time of its generator. 

- When muxing multiple TS, the MUX shall use the PCR from the multiple inputs to determine the final bitrate (and add null packets if a target bitrate is specified) and recompute the PCR values based on the new theoretical bitrate.
- In broadcast, the PCR are recomputed by the modulator with a great accuracy. This is a wall clock example. 
- When an encoder generates a TS file, the production of packets is done at the speed  of the CPU or the output disk (whichever is slower). In that case, the PCR are not based on the wall clock, they are based on the theoretical bitrate of the stream.

So, in the general case, for a generic tool such as TSDuck, the PCR from the stream is a more reliable source than the wall clock.

And remember my previous comment: using the wall clock time is not compatible with the wording of `avt-rtp-mp2t-00.txt` while using the PCR is.

lelegard, 2019-08-05:
Version 3.18 has been released and includes RTP support. Please open a new issue in case of problem with it.

closed 2019-08-05 by lelegard