tsduck/tsduck #320: Save TS file at GOP boundary

felipewd, 2019-06-13:
Hi,

I've been trying to achieve the following scenario:

1) Mux is receiving a continuous DVB-T stream;
- Read 1 DVB-T from another and record it in file;
3) Use to stream it the mux, switching from the continuous one to the file (and then back to the original source);

(1) and (3) are being achieved using tsswitch, and as the second source is just a local port waiting for the stream to arrive (and using "next" at the control port to input it). When the file ends it switches back to the input.

But in order for (2) to be achieved with clean boundaries for smooth transition, it needs to start and end with a GOP. Presently, recording and input with `-P until --duration --seconds 3600` as described in section 5.2.1 of the user's guide. As far as I can see, it doesn't guarantee neither.

I saw the discussion on issue #202...not sure if that's the case. 

One thought I had was to use some tool such as ffmpeg to make sure of this, but since the `-f mpegts` output is really wrong for DVB-T, I'd have to save the tables from the original file, and then replace the tables generated by ffmpeg with the original ones. In your experience is that the way to go?

Thanks.

lelegard, 2019-06-13:
The situation you describes looks like video splicing, which is a notoriously difficult problem. Ad insertion using SCTE-35 cues is a use case and, while manipulating cues is relatively easy, properly splicing the audio and video is complicated at system level.

TSDuck processes generic continuous streams, not disk files where you can seek back and forth. In this context, cutting to GOP boundaries is not so easy. In broadcast streams, a GOP is usually a PES packet. So, they are easy to locate. But they are not easy to properly cut.

In your case, you need to cut the first part of the file, up to the start of the first GOP. Then, you need to cut the end of the file, right after the last complete GOP. Am I right?
 
First, it is easy to drop packets up to the start of the first video PES packet. However, what about audio? If the first audio PES packet starts before the first video PES packet, then you missed part of the audio. You may want to let audio pass at the first PES packet and then wait for the first video PES packet. But this does not work either. You may in fact start on the _previous_ audio PES, just after having missed the start of the previous video PES.

Then, what about the end? When you encounter the end of the stream, you realise that you should have cut after the end of the previous PES packet. Bit, oops, it's too late, you let the start of the incomplete PES packet pass and there is no way to revert the TS packets which are already gone.

So, before implementing something, either extending an existing plugin or creating a new one, we have to specify what should be done in terms of TS packet processing. And, in that case, this is not as easy as it seems.

felipewd, 2019-06-13:
Hi @lelegard, thanks for the quick answer.

Well, this certainly looked a lot easier a little while ago :-)

Since the saved-file can be processed before it's  'ready' to be streamed, I thought maybe using ffmpeg with '-c copy' to preserve the codecs but just cutting everything before the first keyframe (and same thing for the last) would work. Didn't realise this would be a problem for audio.

I could re-encode the whole thing, if you'd think it'd be easier...just not sure what to do to preserve all the PID tables and info required to generate a proper DVB-T output (ffmpeg isn't very good at all with this).

As far as the actual splicing goes, I'm working under the assumption that using something like this:

`tsswitch --infinite  --receive-timeout 2000 --remote 44444 -I ip 239.0.0.2:1010 -I ip 12345 -O 239.0.0.3:1020` 

would work, because if the program previously recorded is scheduled to be streamed at 10:00PM today, all I had to do would be at the scheduled time:

`tsp  -I file ./recorded.ts -O ip 127.0.0.1:12345` and then `echo 1 > /dev/udp/127.0.0.1/44444` .

Then as soon as the video ended, it would go back to the first multicast address, which is non-stop content. But now I'm thinking the issue is not _just_ how to proper cut/re-encode the video, am I right?