tsduck/tsduck #321: Request: Merge-Tables plugin

lars18th, 2019-06-14:
Hi,

Based on the idea of #318 I do some DIY test merging 188 packets interlaced from two streams (with identical CBR). Then I marked them with label 1 and 2 using the regular packet position (easy: `1212121212...`).

However, now I need to apply the same functionality as the `merge` plugin, but this plugin only works with two streams: the main one and another from the standard input. So, in this scenario we need another plugin with similar (or identical functionalities) but based on pids/flags on the same TS.

I deep describe the scenario:
- Inside the same TS are 0-21 pids from two different streams.
- Each table is marked with an unique flag (1 or 2), so pid 0 from TS-1 has flag 1 and pid 0 from TS-2 has flag 2.
- Other pids (data) have flags too, but without collisions (unique pids for each substream). 
- Then the idea it's something like `tsp -I file input.ts -P merge-tables main-flag=1 merge-flag=2 -O file output.ts`.
- And the rest of the plugin options can be the same.

You think this has sense?
Or is there an alternative?

lars18th, 2019-06-17:
Hi,

More or less, I feel I can do that with something like:
```
tsp ... -P sections -n -s --pid 16 --pid 8016 -P sections -n -s --pid 18 --pid 8018 ...
```

And this "merges" the tables 16 (SDT) and 18 (EIT) with the alternatives at 8016 and 8018.

However, one simple question: With the plugin `sections` it's possible to accomplish the **entire** functionality of the `merge`plugin (regarding the PSI/SI tables)?

Thank you!

lars18th, 2019-06-18:
Hi @lelegard ,

As commented in the #325 it's easy to rename multiple sub-streams (original TS muxed manually inside the same TS). When every sub-stream has the same ID, then it's more easy to adapt the PSI/SI tables.

However, despite the EIT tables (we can use the `sections` plugin for this task) we don't have any method to combine (or join) multiple PAT, SDT/BAT or NIT tables. So I suggest a new plugin to do something like (sub-streams have labels 1 and 2):
```
tsp ... -P remap --only-label 2 -n -u 0-21=8000 \
 -P pat-join -p 8000 \
 -P nit-join -p 8016 \
 -P sdt-join -p 8017 \
 -P sections -n --pid 18 --pid 8018 \
 ...
```

Or perhaps a simple extension of the `sections` plugin with `--pat`, `--nit`, `--sdt` and `--bat` options; or any similar solution. But in any case I feel a TABLE JOIN it's a must have.

You agree?

lelegard, 2019-07-02:
See the new plugin `psimerge` and example 5.2.26 in the user's guide.

lars18th, 2019-07-03:
> See the new plugin `psimerge` and example 5.2.26 in the user's guide.

Thank you @lelegard !
I'm testing it, and it seems to work. :+1:

lars18th, 2019-07-03:
Hi @lelegard ,

After some tests I have these comments about this plugin:
- I suggest to filter out of the TOT/TDT tables from the merged stream; or almost comment it in the documentation to include before the `psimerge` a `filter` plugin to remove them. If not, these tables will overlap.
- When merging the EIT tables, the events of the merged programs aren't linked to the target service. I resolve this problem incorporating a previous plugin for changing the TSID and the oTSID. For example: `tsp -I file --interleave=1 --label-base 11 file1.ts file2.ts -P tsrename --only-label 11 -t 1 -o 0x2200 -P tsrename --only-label 12 -t 1 -o 0x2200 ... -P psimerge --main-label 1 --merge-label 2 ...` and then before the `psimerge` both streams will have the same ID and the EIT merge process generates correct outcomes.

I hope this helps.
Regards.

lelegard, 2019-07-04:
The TDT/TOT are now passed from the main stream only. The TDT/TOT from the merge stream are dropped. The option `--time-from-merge` can be used to do the opposite.

The EIT Actual from the merge stream are now patched with the main TS id. Their original network id is left unmodified.

lars18th, 2019-07-05:
Hi @lelegard ,

Regarding the last updates...
1. The EIT from the merge stream now works as expected! :+1: 
2. The change regarding the TDT/TOT tables is good. However still a problem: The Network Synchronization in the pid 21 (0x15) isn't considered. I suggest to block it from the merge stream, or advise in the documentation to filter out it before the `psimerge`.

Thank you for this unvaluable project! :+1:

lelegard, 2019-08-05:
Version 3.18 has been released.

closed 2019-08-05 by lelegard