tsduck/tsduck #322: Problem: reduce plugin and bitrate

lars18th, 2019-06-14:
Hi,

When using the `reduce` plugin the bitrate calculated by the TSDuck doens't change. That's the expected behaviour?

Futhermore, I ask why exists the `reduce` plugin and not the `expand` plugin? I know about the global `-a|--add-input-stuffing`. However, this can only be used at the start, and not inside a processing chain.

Any ideas?

lars18th, 2019-06-14:
> Futhermore, I ask why exists the `reduce` plugin and not the `expand` plugin? I know about the global `-a|--add-input-stuffing`. However, this can only be used at the start, and not inside a processing chain.

Regarding this, in the documentation:
```
Section: tsp
Subsection: Modifying, inserting and deleting packets

The last case cannot be directly implemented. To achieve data injection, 
a plugin usually “steals stuffing”. Each time a new TS packet needs to be 
injected, a plugin waits for the next null packet (i.e. a packet in PID 0x1FFF) 
and replaces this null packet with the new packet to insert.
```

So, don't care. It's impossible!

But I need some info about "retarget" the Bitrate after the `reduce` plugin.
Please...

lars18th, 2019-06-14:
Hi,

To check the problem execute:
```
$ tsp -a 1/1 -I file file.ts -P reduce 1 2 -P analyze -O drop

$ tsp -I file file.ts -P analyze -O drop
```

And compare the results.

Any solution to set the correct bitrate **after** the `reduce` plugin?

lelegard, 2019-06-14:
The option `-a 1/1` adds 1 null packet after every 1 input packet. So, the result is 50% null packets interleaved with 50% input packets.

The plugin `-P reduce 1 2` removes an average of 1 null packet every 2 packets. So, it removes 50% of the stream (provided that there are enough null packets, which is the case here since we already have 50% null packets).

So, the `reduce` plugin exactly removes what `-a 1/1` added and the two commands are consequently identical.

lars18th, 2019-06-14:
Yeah, that's why I wrote this dumb example.
The problem is about the resultant bitrate... I'm not sure if the correct value for the bitrate is used _after_ the `reduce` plugin.

I'll investigate more.

lelegard, 2019-06-14:
Maybe there is a misunderstanding between the "user-specified" bitrate and the "estimated bitrate". In the `tsanalyze` command, you can specify a `--bitrate` option on the command line. This is the "user-specified" bitrate. It may be correct and completely incorrect. The same option is also available with `tsp`. When it is not specified, then the initial bitrate is evaluated on input, based on PCR's.

This is illustrated in the following three examples.

First, the simple one. There is no user-specified bitrate. The bitrate is evaluated at input. It is also evaluated in the `analyze` plugin, giving similar results.
~~~
$ tsp -I file tsduck-test/input/test-001.ts -P analyze -O drop
...
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ......................... 47,802,489 b/s   51,870,785 b/s  |
|  Estimated based on PCR's: ............... 47,790,424 b/s   51,857,694 b/s  |
|-----------------------------------------------------------------------------|
...
~~~

Here, we apply the two modifications which cancel each other. On input, the evaluation is adjusted with `-a 1/1`, giving 95 Mb/s. This is the "user-specified" bitrate on input. But the `reduce` plugin reduces it back to the same value.
~~~
$ tsp -a 1/1 -I file tsduck-test/input/test-001.ts -P reduce 1 2 -P analyze -O drop
...
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ......................... 95,604,980 b/s  103,741,574 b/s  |
|  Estimated based on PCR's: ............... 47,790,424 b/s   51,857,694 b/s  |
|-----------------------------------------------------------------------------|
...
~~~

Here, as an example, we provide some stupid "user-specified" bitrate: 2 Mb/s, becoming 4 Mb/s because of `-a 1/1`. You can see that the user-specified bitrate is correctly reported but the evaluated bitrate is still exactly the same.
~~~
$ tsp --bitrate 2,000,000 -a 1/1 -I file tsduck-test/input/test-001.ts -P reduce 1 2 -P analyze -O drop
...
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: .......................... 4,000,000 b/s    4,340,425 b/s  |
|  Estimated based on PCR's: ............... 47,790,424 b/s   51,857,694 b/s  |
|-----------------------------------------------------------------------------|
...
~~~

So, everything looks good to me.

lars18th, 2019-06-14:
Hi @lelegard ,

Thanks for the analysis. I'm sure your opinion is true.

However, I have this problem: I need to use the `reduce` plugin to _perfect_ reduce the bitrate to the value for specific DVB-T transmission (using the HiDes modulator). And I don't know how to reduce to an specific value _and_ then remark the PCRs.

Any idea to accomplish this task?
Thank you!

lelegard, 2019-06-14:
When the output is a "serious" modulator, it should produce the exact bitrate corresponding to the modulation parameters, generating null packets when necessary. So, the idea is usually to slightly underfeed the modulator to be sure to never overflow it and let the modulator (or its driver) produce one null packet from time to time to compensate the small underflow.

This works with a Dektec modulator, for sure. I do that all the time.

I have not tested with the HiDes modulator. Maybe it works, maybe not. If it does not, then, well, we just have to conclude that having something ___perfect___ (your word) with a 169$ modulator and free software is not possible.

lars18th, 2019-06-15:
Hi @lelegard ,

I understand your comment. However, the problem is not the modulator quality, or the modulator functionality. Let me to explain a bit more:

- Starting from a TS with a high CBR value (30~40Mbps) that's the composition (interlacing) of other captured TS, I want to broadcast it with a DVB-T modulator.
- Then it's required to remove unused services and NULL stuff to reduce the _CBR_ to the _exact_ value of the tunning parameters. And I can do that with the `reduce` plugin.
- For sure the target bitrate needs to be _slightly_ less than the target bitrate. And all DVB-T modulators can compensate this problem inserting NULL stuff. Otherwise the modulator would be useless.
- But the problem are the PCR values. Using the `pcrbitrate` plugin the timestamps are recomputed based on the TS _real_ bitrate. And for this reason the real bitrate needs to be the _target_ bitrate.

So, the problem is this: how to use the `reduce` plugin (or any other plugin combination) to reduce the bitrate to a target bitrate?

I'm sorry for using the word **perfect**, as I only want to say **correct** bitrate. Without the _correct_ bitrate the PCR marks will be incorrect and then the TS feeded to the modultor will have troubles at the player side.
Any idea to overcome this problem?

Thank for this amazing opensource project!

lars18th, 2019-06-17:
Hi @lelegard ,

Before making more comments on this topic, I need to say that I'm sure there is an error with the bitrate and `reduce` plugin. I'm doing this:

- Capture one TS from a DVB-T tuner. Then remove some services and "reduce" the bitrate. Finally reconstruct some PSI/SI tables and remark PCRs. More or less like:
- `tsp ... svremove -a -s ... -P reduce 8 40 -P pcradjust -O file out.ts`.
- But when executing `tsanalyze.exe -w out.ts` I see different values depending on the `pcradjust` presence.
- Without `pcradjust`: Estimated based on PCR's: ... 31,649,798 b/s
- With `pcradjust`: Estimated based on PCR's: ... 39,811,764 b/s

So, I feel the bitrate that uses the `pcradjust` it's wrong, as the `reduce` plugin is in fact changing it. But I don't see any reference to this value in the source code.

Any solution to this problem?

lars18th, 2019-06-18:
Hi @lelegard ,

At time I use this workaround for this problem:
`tsp ... -P reduce 8 40 -O fork 'tsp --bitrate 31668449 -P pcradjust -O file out.ts'`

I put it here only as a reference. However, I want to know your opinion about this issue: The `reduce` plugin needs to recalculate the bitrate or not?

lelegard, 2019-06-21:
This cannot work this way. There is a loop in the references.

PCR's are timestamps. I mean _real wall clock time_. The plugin `pcradjust` shall be used when there is some significant jitter or when the PCR's are completely out of sync (after a loop for instance). Here, your original PCR's are correct, they have the right wall clock time. If the removal of packet is regular, there is no need to change them. If the removal of packet is not regular, because of an unbalanced distribution of null packets, then there is only some jitter to fix. 

But you cannot let the system determine the bitrate based on PCR's and then adjust the PCR's based on this value, computed from the same PCR's. In other words, you cannot use one source as a reference to modify the same source.

PCR adjustment shall be done from a reliable reference. In your case, 1) you know the input bitrate because it comes from a DVB modulation, 2) you know exactly the removal ratio from your instance of `reduce`. So, you know the resultant bitrate. And this is your only reliable reference.

Consequently, you _must_ use `-P pcradjust --bitrate ...`. And you already found it....

lars18th, 2019-06-21:
> Consequently, you _must_ use `-P pcradjust --bitrate ...`. And you already found it....

Yes, that's the solution. However, another time I suggest a simple warning in the documentation: "This plugin doesn't change the internal bitrate value, although it does change the TS bitrate."

After said that:
> If the removal of packet is not regular, because of an unbalanced distribution of null packets, then there is only some jitter to fix.

It's just for that! After muxing multiple CBR TS with VBR streams, and reduce the bitrate to a target value (for sending it to a modulator), _all_ PCR marks will be incorrect. That's true in every case! So the use of the `pcdadjust` plugin it's mandatory. And the problem is quite simple: the `pcradjust` uses the bitrate value, but the `reduce` plugin doesn't change it (the value, not the real bitrate). So a the command `tsp ... -P reduce ... -P pcradjust ...` will do nothing everytime.

And as you pointed the solution it's the `--bitrate` prameter _**of**_ the `pcradjust` plugin. So for this reason I suggest to document it. :wink: 

Regards.

lars18th, 2019-06-25:
Hi @lelegard ,

I have an idea for a precise bitrate reduction. The reason is the _deterministic_ muxing of multiple TS (that's the 1-to-1 interleaving described at https://github.com/tsduck/tsduck/pull/331#issuecomment-504969384):

- Include in the `reduction` plugin the option `--target-bitrate X,Y` where `X` is the target bitrate and `Y` is the steps to do it.

And the algorithm is:

1. Z := bitrate; (current)
2. T := Z - X; (always the target bitrate is less than current bitrate) (T is the number of packets to remove)
3. repeat Y times:  (the objective is for Z packets remove T packets)
4. ........     ts := T / Y;  (integer division)
5. ........     zs := Z / Y;  (integer division)
6. ........     REMOVE(ts,zs);  (remove ts packets from zs packets)
7. ........     T := T - ts;
8. ........     Z := Z - zs;
9. ........     Y := Y - 1;
10. end

So, if the `step` value is sufficient high it's possible to reduce from _any_ bitrate to _any_ bitrate.

What you think?

felipewd, 2019-07-01:
hi @lars18th,

Thanks for the answer. The file I'm using for testing is a SPTS, so the 0x1 service I'm removing and replacing it with `null` packets/should be enough for merge, shouldn't it?

Specially, in the case of the testing case, I'm removing the `0x1` service and merging with the same input (essentialy replacing it with itself).

closed 2019-07-31 by lelegard