tsduck/tsduck #323: Question: How to change the Bitrate in the NIT table?

lars18th, 2019-06-14:
Hi,

I need to update the bitrate manifest inside the NIT table of a current TS.
Any plugin to execute this task?

Thank you!

lelegard, 2019-06-14:
What is the "bitrate manifest" in the NIT?

lars18th, 2019-06-14:
> What is the "bitrate manifest" in the NIT?

The _Terrestrial Delibery System_ Descriptor (0x5A) inside the NIT table. I need to update this descriptor after using the `reduce` plugin.
Any idea?

lelegard, 2019-06-14:
Ah, ok, you want to update the modulation parameters (which, in turn, determine the bitrate, this is correct).

The plugin `nit` provides a few specific on-the-fly transformations on the NIT, but not that one specifically. For any generic modification like this, the process is:

- Extract the tables from the PID in XML format using the plugin `tables`.
- Manually modify the XML file (this is text). Here, you can do what you want, really.
- Reinject the tables in the stream using the plugin `inject`, replacing the previous content of the PID.

In the case of the NIT, just be careful to extract all tables from PID 16, namely the NIT Actual and all NIT Other (if there are any).

lars18th, 2019-06-17:
> Ah, ok, you want to update the modulation parameters (which, in turn, determine the bitrate, this is correct).

YES!

> The plugin `nit` provides a few specific on-the-fly transformations on the NIT, but not that one specifically. For any generic modification like this, the process is:
>
> * Extract the tables from the [...]

It works, thank you! :+1: 
```
tsp -I file input.ts -P filter --pid 16 -P tables --xml-output nit.xml -O drop
nano nit.xml
tsp -I file input.ts -P inject --pid 16 -r --xml nit.xml -O file out.ts
```

However, as a simple question: Any option to have something like this?
```
tsp ... -P nit --replace "terrestrial_delivery_system_descriptor"=file.xml -O ...
```

Now it's easy to edit the XML files (thank you for this unvaluable option). However, instead of replace the full table, I ask about replace part of it. Any change?

lelegard, 2019-06-21:
> tsp ... -P nit --replace "terrestrial_delivery_system_descriptor"=file.xml -O ...

No, because, in this case, you must also specify the TS id. Then, if you extend this to any type of table, the complexity is so high that it becomes useless. Too many structures, too many descriptors. The plugins such as `nit`,  `pmt`, etc. have some options to handle specific use cases on the fly. But there always will be other cases which are not handled. So, the generic method is the XML one.

lars18th, 2019-06-21:
Hi @lelegard ,

Thank you for your explanation. I agree and I accept your opinion on the KISS principle. :+1: 

However, with all due respect I have created this XML file with _only_ this data:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT>
    <transport_stream>
      <terrestrial_delivery_system_descriptor centre_frequency="698,000,000" bandwidth="8MHz" priority="HP" no_time_slicing="true" no_MPE_FEC="true" constellation="64-QAM" hierarchy_information="0" code_rate_HP_stream="2/3" code_rate_LP_stream="1/2" guard_interval="1/4" transmission_mode="8k" other_frequency="true"/>
    </transport_stream>
  </NIT>
</tsduck>
```

And the idea is to _REPLACE_ the used table with this content. I understand it's hard to implement functions/parameters for ALL values (that's impracticable). However, I feel the solution of an XML _replace_ will provide the best from both sides: flexibility and simplicity.

What you think?

lelegard, 2019-06-25:
No way, too specific, impossible to address the general case.

closed 2019-06-25 by lelegard

lars18th, 2019-06-25:
> No way, too specific, impossible to address the general case.

I'm disappointed to hear that.