tsduck/tsduck #324: Bug: tsswitch stuck on remote command

felipewd, 2019-06-14:
Hi!

I've been trying to use tsswitch using the following scenario:

`mkfifo b; tsswitch --delayed-switch  --receive-timeout 2000 --infinite -v --debug    --remote 44444 -I ip 12345 -I file ./b  -O ip 127.0.0.1:12346`

Using a constant source on localhost:12345, and then using a 10-second (3 second because of repetition) file to stream no the FIFO b:

`tsp -I file -r 3 ./3.ts -P pcrbitrate -P regulate -O file ./b`

When I try to change it to 1:

`echo "1" > /dev/udp/127.0.0.1/44444`

I get an eternal output of:

```
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
```

And, after that, all other commands on the remote port are read but ignored:

```
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* received command "0", from 127.0.0.1:38478 (2 bytes)  <====== New command
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
```

This is using version `3.17-1189`.

This is on a linux machine.

If you need any more info, please let me know.

felipewd, 2019-06-14:
Oh, just if it's important...the initial messages before the endless output are:

```
* Debug: file[1]: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: setting socket reuse address to 1
* Debug: binding socket to 0.0.0.0:44444
* Debug: UDP server thread started
* Debug: ip: output thread started
* Debug: ip[0]: input thread started
* Debug: ip[0]: received start request, current: true
* Debug: ip[0]: waiting for input session
* Debug: ip[0]: starting input plugin
* Debug: ip[0]: setting socket reuse address to 1
* Debug: ip[0]: binding socket to 0.0.0.0:12345
* Debug: ip[0]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 0, true
* Debug: file[1]: input thread started
* Debug: file[1]: waiting for input session
* Debug: setting event: WAIT_INPUT, 0, false
* received command "1", from 127.0.0.1:48409 (2 bytes)   <=== Trying to change to FIFO
* Debug: switch input 0 to 1
* Debug: enqueue action SUSPEND_TIMEOUT, 0, false
* Debug: enqueue action START, 1, false
* Debug: enqueue action WAIT_INPUT, 1, false
* Debug: enqueue action SET_CURRENT, 1, false
* Debug: enqueue action NOTIF_CURRENT, 1, true
* Debug: enqueue action RESTART_TIMEOUT, 0, false
* Debug: enqueue action ABORT_INPUT, 0, false
* Debug: enqueue action STOP, 0, false
* Debug: enqueue action WAIT_STOPPED, 0, false
* Debug: executing action SUSPEND_TIMEOUT, 0, false
* Debug: executing action START, 1, false
* Debug: file[1]: received start request, current: false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: file[1]: starting input plugin
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
* Debug: executing action WAIT_INPUT, 1, false
* Debug: not ready, waiting: WAIT_INPUT, 1, false
```

felipewd, 2019-06-14:
Ok, so, the issue exists, but as I first reported it.

The FIFO was not on the same directory, so that's on me :-)

BUT...since I started tsswitch with `--receive-timeout 2000` I expected the timeout to pass and then return it to the input 0 (no actual changes because of `delayed-switch`.

During those 2 seconds I can understand ignoring commands, but not after, right?

lelegard, 2019-06-16:
Hi,
This is a bug which occurs when `--delayed-switch` is used with  `--receive-timeout`.
Thank you for reporting it. It will be fixed.

lelegard, 2019-06-21:
Fixing the problem of the interaction between `--delayed-switch` and `--receive-timeout` seems more complicated than anticipated. I may be obliged to rewrite part of the `tsswitch` core since the VM model that was used may be invalidated (or become too complicated) and I may need to switch to a state machine.

But, in the meantime, note that named pipes are probably not the best way to communicate with `tsp` or `tsswitch`. See my comment in #327.