tsduck/tsduck #327: Bug: tsswitch ignores --primary after 1st event

felipewd, 2019-06-17:
Hi,

Testing the following scenario:

```
mkfifo a b;
tsswitch -v --debug --infinite  -p 1  -I file a -I file b -O ip 239.0.1.2:1234
```

Then steps to show the problem:
1) Start endless signal on FIFO `a`;
2) Start a 10s TS on FIFO `b`;   < === Switch happens;
3) After ending, start again a 10s TS on FIFO `b`; <== Doesn't switch anymore.

Using `--infinite` implies (in my understanding) that `-p 1` (2nd entry) should always come up when it becomes available, right?

felipewd, 2019-06-17:
Also, I just noticed:

When the switch happens, the 1st FIFO is actually closed:

```
* Debug: file[1]: input thread started
* Debug: file[1]: waiting for input session
* Debug: file[1]: starting input plugin
* Debug: file[0]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 0, true
* Debug: setting event: WAIT_INPUT, 0, false
========= <==========  Starts 2nd input

* Debug: file[1]: input plugin started, status: true
* Debug: setting event: WAIT_STARTED, 1, true
* Debug: setting event: WAIT_INPUT, 1, false
* Debug: enqueue action SUSPEND_TIMEOUT, 0, false
* Debug: enqueue action NOTIF_CURRENT, 0, false
* Debug: enqueue action SET_CURRENT, 1, false
* Debug: enqueue action NOTIF_CURRENT, 1, true
* Debug: enqueue action RESTART_TIMEOUT, 0, false
* Debug: enqueue action ABORT_INPUT, 0, true
* Debug: enqueue action STOP, 0, false
* Debug: enqueue action WAIT_STOPPED, 0, false
* Debug: executing action SUSPEND_TIMEOUT, 0, false
* Debug: executing action NOTIF_CURRENT, 0, false
* Debug: executing action SET_CURRENT, 1, false
* Debug: executing action NOTIF_CURRENT, 1, true
* Debug: executing action RESTART_TIMEOUT, 0, false
* Debug: executing action ABORT_INPUT, 0, true
* Debug: executing action STOP, 0, false
* Debug: file[0]: received stop request
* Debug: executing action WAIT_STOPPED, 0, false
* Debug: not ready, waiting: WAIT_STOPPED, 0, false
* Debug: executing action WAIT_STOPPED, 0, false
* Debug: not ready, waiting: WAIT_STOPPED, 0, false
* Debug: file[0]: stopping input plugin
* Debug: input 0 completed, success: true
* Debug: setting event: WAIT_STOPPED, 0, false
* Debug: executing action WAIT_STOPPED, 0, false
* Debug: clearing event: WAIT_STOPPED, 0, false
* Debug: file[0]: waiting for input session
```

This causes the following input entry to close:

`tsp -I ip 239.1.2.3:1020 -P pcrbitrate -P regulate -O file ./a`

Which I think is also not right. Could you confirm?

lelegard, 2019-06-21:
This is a side effect of Unix named pipes. This type of device does not behave just like a permanently established pipe between applications. Each time a writer exits, the readers get an end-of-file.

Named pipes are not always appropriate for all usages. Depending on what you want to do, the input plugin `fork` could be a better - and portable - alternative.