tsduck/tsduck #328: question: --terminate equivalent using -P merge

felipewd, 2019-06-17:
Hi,

This is an awesome project! I'm throughly enjoying testing it with our scenario.

Using `mux` plugin I can use `--terminate` to indicate the packet processing should terminate after the input ends.

Is there any way to achieve this using `merge`? 

I'm trying to merge a 10min-TS file in a multicast live stream...but would like tsp to quit after the input file ends. Can this be achieved?

Thanks!

felipewd, 2019-06-17:
One option would be to use `-P until --seconds <time of the input>`  after `merge` before `-O`...but I'm hoping there's a more elegant solution :-)

lelegard, 2019-06-21:
`merge --terminate` will be added.

lelegard, 2019-06-24:
Added options `--terminate` and `--joint-termination` to plugin `merge` in commit 6176bee272c4fd9d155fd958109bce722d3dfd87

felipewd, 2019-06-28:
Works great, thanks!

closed 2019-07-31 by lelegard