tsduck/tsduck #329: Unable to start tsduck 3.17 on Debian

nsaadi, 2019-06-18:
Hi,
I use for the moment the version 3.16 all is ok, but when I upgrade to version 3.17 all is ok and the installation is done.
But I have the following error message when i try to use any binary:
_user@debian:~$ tsversion --help
tsversion: /usr/lib/x86_64-linux-gnu/libcurl.so.4: version `CURL_OPENSSL_4' not found (required by /usr/bin/tsduck.so)
tsversion: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.27' not found (required by /usr/bin/tsduck.so)_
Can you please help us to fix this error.
Thx

apallier, 2019-06-19:
Hi,
I have the same problem on the following Debian version (8 Jessie):

```
root@debian-test:~# uname -a
Linux debian-test 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1+deb8u2 (2017-03-07) x86_64 GNU/Linux

root@debian-test:~# hostnamectl
  ...
  Operating System: Debian GNU/Linux 8 (jessie)
            Kernel: Linux 3.16.0-4-amd64
      Architecture: x86-64
```

I think the problem comes from a too old "libc6" library => 2.19 instead 2.27

```
root@debian-test:~# aptitude show libc6
Package: libc6
...
Version: 2.19-18+deb8u7
...
```

lelegard, 2019-06-21:
Do you use the pre-built `.deb` files from GitHub or tsduck.io?

These packages are built on Ubuntu and may not work on Debian when the Debian version is behind Ubuntu in terms of libraries. Between the releases of TSDuck 3.16 and 3.17, Ubunto moved from libcurl 3 to libcurl 4. If your Debian version is still with libcurl 3, then a dependency is broken.

Note that this is a binary dependency, not a source code dependency. You can rebuild TSDuck from source as [explained here](https://tsduck.io/doxy/building.html). This is quite simple. Just use `apt install` on prerequisite packages (if any is missing) and then `make deb` in `tsduck` source directory to build new `.deb` packages for your system.

lars18th, 2019-06-21:
My two cents:

TSDuck it's quite easy to compile (compared to other large projects), so I suggest to compile it in your Linux machine. It's the easiest way to use it!

Regards.

closed 2019-06-23 by lelegard

nsaadi, 2019-06-24:
Hi all,

I rebuild it with our Debian version, and it's work fine.

Thx