tsduck/tsduck #330: Question: Remove first NULL stuff of a TS

lars18th, 2019-06-18:
Hi,

I have one question to all:

- I'm doing some processing like this:
```
 tsp -I ... -P ... | tsp --bitrate NNNNN -P pcradjust -O ...
```
- The problem is that the first `tsp` process generates some NULL stuff at the begining. And I want to remove these NULL packets.

So, it's possible to remove _initial_ NULL packets? Or initiate the processing when the "first not null" appears in the stream?

I hope you can help me with some solution.
Thank you!

lelegard, 2019-06-21:
I don't see an obvious solution for now.

I can propose to add options `--set/reset-permanent-label` to plugin `filter` in addition to the existing `--set/reset-label` options. The now options would set/reset labels on _all_ subsequent packets, not only selected ones.

Your case would be addressed by:
~~~
tsp -I ... \
    -P filter --pid 0x1FFF --set-label 31 \
    -P filter --pid 0x1FFF --negate --reset-permanent-label 31 \
    -P filter --label 31 --negate \
   ...
~~~
Maybe there is a simpler solution.

lars18th, 2019-06-21:
> The nEw options would set/reset labels on _all_ subsequent packets, not only selected ones.

But I feel this `--set/reset-permanent-label` can be done using a simple `-P filter -v --reset-label XX` (almost I do it in this way until now).

However, perhaps I'm not described well the scenario.
The _input_ TS has this structure (example):
`nnnnnnnnnnnnnnnnnn1n2nn34567n8n90ABCnDEFn...`

What I want to do it's block the first null packets. After the first non-null packet (1), the rest is passed.
Any idea?

lelegard, 2019-06-24:
> But I feel this --set/reset-permanent-label can be done using a simple -P filter -v --reset-label XX (almost I do it in this way until now).

No, because the "permanent" options act as triggers while the non-permanent ones have a local impact only.

> What I want to do it's block the first null packets. After the first non-null packet (1), the rest is passed.

This is exactly what is achieved in the example with the option (to come) `--reset-permanent-label`.

lelegard, 2019-06-24:
Done in commit f868bd9ee2564fb3fc01cb100856d0e8d2c2a2f8

lars18th, 2019-06-24:
> Done in commit [f868bd9](https://github.com/tsduck/tsduck/commit/f868bd9ee2564fb3fc01cb100856d0e8d2c2a2f8)

Thank you! I check it soon.

lars18th, 2019-06-24:
Hi @lelegard ,

It works! Now I can use `... -P filter -p 8191 -n --set-permanent-label 1 -P filter -l 1 ...` to remove these NULL packets at start.

Thank you!
You can close the issue.
Regards.

closed 2019-06-24 by lelegard