tsduck/tsduck #332: Request: duplicate any when using labels

lars18th, 2019-06-20:
Hi @lelegard ,

Using the `--only-label` parameter with the `duplicate` plugin I feel it's good to have something like:
```
tsp ... -P duplicate --only-label 1 --all 0=100 ...
```

That's represents an unconditional (`-a` or `--all`) duplication to the pid 100 for all packets with the `label` 1.

You agree?

lelegard, 2019-06-21:
I don't understand the question. Could you specify it precisely in terms of packet processing?

lars18th, 2019-06-21:
> I don't understand the question. Could you specify it precisely in terms of packet processing?

I'll try:

- Mark with `-P filter --set-label 1...` different packets **despite** the pid used.
- COPY with `duplicate` the packets with the `label 1`.

How to do that?

lelegard, 2019-06-21:
So, it is the same issue as #333 ?

lars18th, 2019-06-21:
> So, it is the same issue as #333 ?

Absolutely not!

Please, let me to know how to "duplicate packets with an specific mark"... "maintaining the pid number (and the label if it's possible)". Obviously without inserting 8190 times the same plugin.

I can't find a way to do it. :cry:

lars18th, 2019-06-21:
Sorry @lelegard ,

I'm quite stupid! :hankey: 

That's the solution (with all to the same pid):
 - `-P duplicate --only-label 1 -u 0-8190=0 --P remap --only-label 1 -s -u 0-8190=XXX`

and this (maintaining the pid):
 - `-P duplicate --only-label 1 -u 0-8190=0`

Thank you for this extremely flexible project! :+1:

lars18th, 2019-06-23:
Hi @lelegard,

I need to reopen this issue as this can't be done:
`-P duplicate --only-label 1 -u 0-8190=0 --P remap --only-label 1 -s -u 0-8190=XXX`

The problem is because the `remap` can't be limited to the duplicated packets, as it's remaping too the original packets. We don't have at all the option to duplicate and change the label!

So, one of these options it's required:
1. Add `--set-label` to the `duplicate` plugin.
2. Add `--all` to the `duplicate` plugin and `--all-and-remap` too.

Personally, I feel `--set-label` is the most elegant and simple.
What you think?

Remember: The objective is to duplicate all packets with a label *and* set all of these packets with the same pid.

Regards.

lelegard, 2019-06-23:
It could be also done by the same `--single` option as in `remap`. Thus, `remap` and  `duplicate` would be more consistent.

I will add `--single` and `--set-label` to `duplicate`.

lelegard, 2019-06-23:
`--single` and `--set-label`added to `duplicate` in commit e61abf233f783623acec23e6d8258be7afb8868d.

closed 2019-06-23 by lelegard

lars18th, 2019-06-24:
Hi @lelegard ,

Thank you very much! I see you're a very reasonable person. And I'm very happy to have the opportunity to use this incredible project. :+1:  :smiley: