tsduck/tsduck #333: BUG: duplicate with only-label

lars18th, 2019-06-20:
Hi @lelegard ,

Please try this test:
```
tsp ... -P filter -p 100 --set-label 1 -P duplicate --only-label 1 100=200 -O drop
```

Then you will obtain a lot of
```
Warning: duplicate: buffer overflow, dropping packet
```

I feel the problem is because the `duplicate` plugin only searchs for NULL stuff packets with label 1, instead of using _any_ NULL packet.

Please, can you check it?
Thank you!

lelegard, 2019-06-21:
This is the normal behaviour. The option `--only-label` is like `--help`, it is a global one, it is managed at `tsp`  level and it is never used by the plugin. With `--only-label`, `tsp` bypasses the plugin for all packets which do not have any of the specified labels. In your case, the plugin `duplicate` never sees any null packet and it cannot duplicate anything.

In your specific example, the command should be:
~~~
tsp ... -P filter -p 100 -p 0x1FFF --set-label 1 -P duplicate --only-label 1 100=200 -O drop
~~~
Thus, the null packets are also tagged with label 1 and are processed by the plugin.

lars18th, 2019-06-21:
> This is the normal behaviour.

OK.

> In your specific example, the command should be:
> ```
> tsp ... -P filter -p 100 -p 0x1FFF --set-label 1 -P duplicate --only-label 1 100=200 -O drop
> ```

Yes. That's the workaround that I found out.

In any case, if this is the correct behaviour, I suggest to insert this example in the documentation. Without labelling the null packets the `duplicate` fails when using the `--only-label`.

Regards.

lelegard, 2019-06-21:
Well, not all specific cases can be individually documented. With a very generic toolbox such as TSDuck, the documentation has to be very accurate on each point. Then, some analysis and synthesis are required from the user.

In the documentation for the `duplicate` plugin, we have this:
> This plugin duplicates the content of several PID’s into new PID’s. The duplicated packets are created by replacing existing null packets. The input stream shall consequently contain at least as many null packets as packets to duplicate.

And this:
> `--only-label label1[-label2]`
> Invoke this plugin only for packets with any of the specified labels. Other packets are transparently passed to the next plugin, without going through this one.

So, it says that duplication needs null packets and that packets without the required labels are not passed to the `duplicate` plugin. So, if you don't pass stuffing to ` duplicate` or any other plugin which inserts packets by replacing stuffing, then the duplication or insertion will never takes place. This isn't that complicated.

lars18th, 2019-06-21:
> This isn't that complicated.

Sorry, not for me!

If you approve, I suggest this simple sentence: "Note: When using this plugin with the --only-label _only_ null packets with this label are replaced."

That's all! In any case, I write here for reference to other people.
Regards.

lars18th, 2019-06-21:
Sorry @lelegard ,

I'm quite stupid! :hankey: 

That's the solution (with all to the same pid):
 - `-P duplicate --only-label 1 -u 0-8190=0 --P remap --only-label 1 -s -u 0-8190=XXX`

and this (maintaining the pid):
 - `-P duplicate --only-label 1 -u 0-8190=0`

Thank you for this extremely flexible project! :+1:

closed 2019-06-23 by lelegard

lars18th, 2019-06-24:
Hi @lelegard ,

Thank you for adding the option `--single` to the `duplicate` plugin in commit https://github.com/tsduck/tsduck/commit/e61abf233f783623acec23e6d8258be7afb8868d. :+1: