tsduck/tsduck #335: Question: --max-input-packets being ignored on pipe input

felipewd, 2019-06-21:
Hi @lelegard,

I've been trying to use `fork` input/output (even processing) to better interact with `tsswitch` such as described on issue #327 , but I hit an annoying problem. 

Using a 800Kbps CBR file for testing, `tsp` is "stuck" for quite a while like this (I'm posting a non-network-related command to demonstrate the issue)

```
# Takes over 47 seconds to play
tsp -I file ./test-800Kbps.ts -P regulate | tsp -r -O play
```

The issue kind of goes away if I use ` --buffer-size-mb 1` such as. File plays after ~4s:

```
# Takes a little over 4 seconds to play
tsp -I file ./test-800Kbps.ts -P regulate | tsp -r --buffer-size-mb 1 -O play
```

So the issue to me seems to be buffer-related. So the next options I tried to play with was reducing the `--max-input-packets 10` makes no difference to the first attempt (still takes >47 seconds to play).

`-P regulate` is working as expected, because this test:

```
# Takes a little over 4 seconds to play
tsp -I file ./test-800Kbps.ts -P regulate | cat > a`
```
Writes about 100KB of data, so it is roughly 800Kbps as the input dictates.

`--debug=2` confirms that the options are being correctly used...not sure what I'm doing wrong here:

```
* Debug[2]: debug level set to 2
tsp: Debug: trying to load /usr/local/Cellar/tsduck/3.17-1189/bin/tsplugin_file.so
tsp: Debug: trying to load /usr/local/Cellar/tsduck/3.17-1189/bin/tsplugin_play.so
tsp: Debug: symbol tspNewInput not found in /usr/local/Cellar/tsduck/3.17-1189/bin/tsplugin_play.so
tsp: Debug: symbol tspNewProcessor not found in /usr/local/Cellar/tsduck/3.17-1189/bin/tsplugin_play.so
tsp: Debug: realtime: yes, using --max-input-packets 10 --max-flushed-packets 1,000
* Debug[2]: file: debug level set to 2
* Debug[2]: play: debug level set to 2
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes

========= > 47 seconds later <============

* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 810,045 b/s
* play: using media player command: "/Applications/VLC.app/Contents/MacOS/VLC" - --play-and-exit
* Debug: play: creating process ""/Applications/VLC.app/Contents/MacOS/VLC" - --play-and-exit"
* Debug: file: input thread started
* Debug: play: output thread started
```

The "initial buffer load" indicating a *lot* of packets makes no sense to me.

Cheers :-)

lelegard, 2019-06-23:
When starting, `tsp` preloads the buffer up to half of its size (8 MB out of 16 by default). This is done to optimise the use of the buffer and to have enough initial PCR's in order to evaluate the bitrate.

For very small bitrates, if you do want to wait that long before starting the processing, you need to reduce the buffer size. For very small bitrates, you do not need a large buffer anyway, all plugins have plenty of time to process the stream without taking the risk to block the input.

The downside is that a small initial load may not give you enough PCR's to evaluate the bitrate. If the plugins you use do not need any bitrate knowledge, this is not an issue. But if any plugin needs to know the TS bitrate (to compute a table insertion interval for instance), then you must know in advance the TS bitrate and specify it on the command line.

lelegard, 2019-06-23:
And the `--max-input-packets` option is used to limit the size of each read operation on the input plugin. It mainly applies _after_ initial buffer loading. To handle very low bitrate streams, it is a good idea to reduce it.

felipewd, 2019-06-23:
That all makes sense. Thanks for the clarification!

closed 2019-06-23 by lelegard