tsduck/tsduck #336: BISS 2 mode CA EMM - table_id is listed wrong (cosmetic)

nautilus7, 2019-06-22:
Hi, in most crypto systems, ECMs have table_id 0x80 or 0x81. In BISS2 mode CA, only possible table_id for the ECM is 0x80, while table_id 0x81 and up are used for the EMMs.

You can find the spec here (page 16): https://tech.ebu.ch/docs/tech/tech3292s1.pdf

As a result, when running tsanalyze for a biss2 mode CA sample, the listing is wrong:

```
===============================================================================
|  TABLES & SECTIONS ANALYSIS REPORT                                          |
|=============================================================================|

.
.
.

|=============================================================================|
|  PID: 0x0020 (32)                          European Broadcasting Union ECM  |
|  Service: 0x0001 (1) TESTCA                                                 |
|-----------------------------------------------------------------------------|
|  0x80 (128, ECM (even)), TID ext: 0x0004 (4)                                |
|      Repetition: ...... 58  ms  Section cnt: ...... 970  First version  19  |
|      Min repet.: ....... 3  ms  Table cnt: ........ 970  Last version:  23  |
|      Max repet.: ..... 118  ms  Versions: ............. 19, 20, 21, 22, 23  |
|=============================================================================|
|  PID: 0x0021 (33)                          European Broadcasting Union EMM  |
|-----------------------------------------------------------------------------|
|  0x81 (129, ECM (odd)), TID ext: 0x0004 (4)                                 |
|      Repetition: ...... 42  ms  Section cnt: .... 1,338                     |
|      Min repet.: ...... 21  ms  Table cnt: ...... 1,338                     |
|      Max repet.: ...... 94  ms  Version: ............ 1                     |
|=============================================================================|
```

lelegard, 2019-06-22:
OK, this is specific to this CAS. I haven't read the full spec, so I do not know how they alternate session words and renew ECM's. In most other CAS, the table ids 0x80 and 0x81 and used to facilitate the section filtering and get notified only once per crypto-period.

Anyway, according to the spec intro, this system is designed to protect contribution feeds, not individual subscribers. So, it is very specific.

nautilus7, 2019-06-22:
Like BISS1, BISS2 will be used for feeds between providers.
Yes, in other CAS, table_ids 0x080 and 0x81 are used interchangeably, but not in BISS2.

The ECM and EMM tables in BISS2 follow the syntax of the private section as defined in paragraph 2.4.4.10 of Rec. ITU-T H.222.0. This is also stated in the pdf I linked in my first post. There is a "version_number" inside every "private section" and thus in every ECM.

tsanalyze detects ECMs with "version_number" 19, 20, 21, 22, 23 (for my sample), so I guess, this is how they "renew" ECMs.

For alternating between odd and even session words (or control words), they use the "transport_scrambling_control" of the each TS packet.

lelegard, 2019-06-23:
OK, their ECM and EMM use the long-section / multi-section syntax while most (all?) other CAS use short-section / mono-section syntax. In that case, the version number is used to trigger a new ECM. Because they conform to the long-section usage rules, TSDuck will get each ECM exactly once. So, this is fine (except the name).

This is strange that they made that choice. On an MPEG/DVB perspective, the versioned long-syntax should be used for cycled tables with a long life-time (e.g. PAT, PMT, NIT, etc) while the short-syntax should be used for small rapidly modified (or even one-shot) sections (e.g. ECM, EMM, TOT, TDT).

That being said, I have seen worse in ATSC where long-syntax tables are changed without changing the version number, which is not even MPEG-compliant...

Anyway, a CAS which publicly publishes the syntax of its ECM's and EMM's must be a joke 😉

rezna11, 2019-06-23:
I think BISS was not meant to be super secure, it’s free and included in
many scramblers and IRDs without extra licenses. It’s only used to make
copyright owners happy. Of course Sky or C+ would never use BISS.

nautilus7, 2019-06-23:
@lelegard BISS2 doesn't seem a joke. Its EMMs are encrypted using asymmetrical encryption (RSA 2048 OAEP). Also the actual audio and video packets are scrambled with DVB-CISSA (AES128 CBC), so they are much more immune to rainbow attacks, unlike CSA. No other CAS does any of these, AFAIK.

@rezna11 we are talking about BISS2 here, which is a step further in security with rolling 128bit control words, while BISS1 uses a constant 48bit control word.

closed 2019-06-23 by lelegard

onajafi, 2025-04-02:
@nautilus7 Can you share your BISS-CA scrambled samples? I've tried to search everywhere but didn't find any .ts file to have such a thing.

nautilus7, 2025-04-02:
https://limewire.com/d/DfWvk#wec1yVrs8X