tsduck/tsduck #337: BUG: Duplicate limited to 193 pids

lars18th, 2019-06-23:
Hi @lelegard ,

Based on https://github.com/tsduck/tsduck/issues/332#issuecomment-504463407
I found this bug:
- OK: `P remap --only-label 1 -n -s -u 0-193=8000`
  - (I use `--only-label` to make sense to the command)
- FAIL: `P remap --only-label 1 -n -s -u 0-194=8000`
  - with this error: `* Error: remap: invalid remapping specification: 0-194=8000`

The problem is because this line:
`ok = ok && pid1 <= PID_MAX && pid2 <= PID_MAX && pid1 <= pid2 && size_t(newpid) + (pid2 - pid1) <= PID_MAX;`
https://github.com/tsduck/tsduck/blob/master/src/tsplugins/tsplugin_remap.cpp#L182

As you can see the number of pids to remap are limited, based on `size_t(newpid)`.
Perhaps this have sense when not using the command: "remap _all_ pids to one".
However, I need this command as I described in https://github.com/tsduck/tsduck/issues/332#issuecomment-504463407.

So, two options:
1. Eliminate this restriction;
2. Or, add one parameter for "remap-all" (and perhaps "duplicate-all" too for `duplicate`).

What you prefer?

lelegard, 2019-06-23:
The bug is not that the remapping is limited to 193 PIDs. The bug is that option `--single` is not taken into account in the PID range check.

Without `--single`, the remapping `0-194=8000` is invalid since the remapped range would be 8000-8194 while the last valid PID number is 8191.

I will fix this.

lelegard, 2019-06-23:
Fixed in commit e61abf233f783623acec23e6d8258be7afb8868d.

closed 2019-06-23 by lelegard

lars18th, 2019-06-24:
> Fixed in commit [e61abf2](https://github.com/tsduck/tsduck/commit/e61abf233f783623acec23e6d8258be7afb8868d).

Thank you!