tsduck/tsduck #340: Error: broken pipe

lars18th, 2019-06-24:
Hi,

I need help with a very strange error:

- I'm using ffmpeg with tsduck with pipes, and I see a runtime error with `tsp` but not with `tsanalyze`:
- OK: `ffmpeg -f mpegts -i file.ts ... -f mpegts | tsanalyze`
- ERROR: `ffmpeg -f mpegts -i file.ts ... -f mpegts | tsp -P analyze -O drop`

The error (from the ffmpeg process) prints:
```
Conversion failed!
Floating point exception
```

And doing some debug with the ffmpeg process the error is related to a BROKEN PIPE. That's the `tsp` has closed the PIPE. For sure, when writing to a file, and reading from that file with `tsp` all works right. Futhermore, this problem doesn't appear with _any_ ffmpeg processing. Just when using some special complex filtering. Almost every time the processing pass, but just in some cases I see this error.

So the conclusion is that a problem exists with the PIPE. But I don't understand why the `tsp` fails, but not the `tsanalyze`.

Futhermore, I need to know if it's possible to solve this problem.
Ideas?

lars18th, 2019-06-25:
Hi,

I'm not sure if this problem can be fixed. After some more debug I see that the problem is related to this:
- In some cases the ffmpeg process "flushes" the output very often. At every "flush" command it writes the entire buffer to the pipe. And when multiple "flush" commands are concatenated, the pipe can be closed because a write error.
- This seems to be a problem of the ffmpeg. However, when using _any_ tsduck tool that uses the simple `tsp::read()` function, or other external tools like `| cat > out.ts` this behaviour don't create any trouble. So I feel the problem is the multi-threading read used by the `ts::tsp::InputExecutor` class. When incrementing the size of the read buffer (with `--buffer-size-mb`) the problem appears after at some point of the stream. Although I can't eliminate the problem.

So, before closing this issue with an "unsolvable" label... Any ideas?

Thank you all for your support!

lars18th, 2019-06-25:
Hi,

I feel I have found a solution:
`ffmpeg -f mpegts -i file.ts ... -f mpegts -max_interleave_delta 20000000 | tsp -P analyze -O drop`

The default `max_interleave_delta` value in ffmpeg is `15000000`, so if you increase this value, the problem disappears.

I hope it helps to others.
Regards.

lelegard, 2019-06-25:
Does `tsp` report any error? `tsp` should not close its input without reason. A good reason to close the input is a corrupted stream, e.g. a packet not starting with 0x47.

> In some cases the ffmpeg process "flushes" the output very often. At every "flush" command it writes the entire buffer to the pipe. And when multiple "flush" commands are concatenated, the pipe can be closed because a write error.

This must be an error in ffmpeg. Writing into a pipe should block, not fail. Or maybe ffmpeg uses some non-blocking form of I/O which fails in case of blocking. Is this on Linux or Windows? In the latter case, you may expect a lot of oddities...

>  So I feel the problem is the multi-threading read used by the ts::tsp::InputExecutor class. 

There is no such thing as a "multi-threading read". All processes are made of threads, at least one, and all threads are created equal. All forms of read execute in a thread. That cannot be the root cause.

lars18th, 2019-06-25:
Thank you @lelegard for your comment!

> Does `tsp` report any error? `tsp` should not close its input without reason. A good reason to close the input is a corrupted stream, e.g. a packet not starting with 0x47.

No. No error report at all!

> This must be an error in ffmpeg. Writing into a pipe should block, not fail. Or maybe ffmpeg uses some non-blocking form of I/O which fails in case of blocking. Is this on Linux or Windows? In the [...]

I test it in Linux and Windows. Identical behaviour. As I've discovered the ffmpeg process writes in the PIPE async with threads, and it closes the PIPE if it chan't write more data _when doing a flush_ !!! So the problem is with the internal ffmpeg logic that triggers a flush when the interlave timming between muxed streams raises the maximum value. We can modify this value with the `-max_interleave_delta` parameter. And using `-f mpegts -max_interleave_delta 20000000` I resolved the problem.

However, I feel it's good to document it.

> > So I feel the problem is the multi-threading read used by the ts::tsp::InputExecutor class.
> 
> There is no such thing as a "multi-threading read". All processes are made of threads, at least one, and all threads are created equal. All forms of read execute in a thread. That cannot be the root cause.

In my first review I found that all TSDuck tools except the `tsp` binary uses a simple loop with the function `tsp::read()` for reading from the STDIN. But the `tsp` tool uses the class `ts::tsp::InputExecutor`. So perhaps this class has a complex loop that blocks at some point, instead of read _every_ time the producer process writes in the PIPE. But that's just my feeling.

Regards.

lelegard, 2019-06-25:
> In my first review I found that all TSDuck tools except the tsp binary uses a simple loop with the function tsp::read() for reading from the STDIN. But the tsp tool uses the class [...]

The `file` input plugin (the one that reads the standard input) is identical to any other TSDuck tool that reads files, at least on a system I/O standpoint. It reads and blocks until some data are read. The plugin reports an error on I/O error, end of file or invalid TS packet not starting with 0x47.

Your diagnosis is very clear: ffmpeg _decides_ to exit when it can't flush. This is probably relevant with text files but a very bad idea with pipes since this class of devices is inherently blocking because of flow regulation.

lars18th, 2019-06-25:
> Your diagnosis is very clear: ffmpeg _decides_ to exit when it can't flush. This is probably relevant with text files but a very bad idea with pipes since this class of devices is inherently blocking because of flow regulation.

Yes. But providing patches for the ffmpeg project is a tedious task. So I'd rather focus on improving TSDuck. :wink:

lelegard, 2019-06-25:
> But providing patches for the ffmpeg project is a tedious task. So I'd rather focus on improving TSDuck. 

Agreed but impossible here. FFmpeg _closes_ its output pipe. So, `tsp` just encounters a quite standard  end-of-file. Can't do anything more, sorry...

lars18th, 2019-06-25:
> > But providing patches for the ffmpeg project is a tedious task. So I'd rather focus on improving TSDuck.
> 
> Agreed but impossible here. FFmpeg _closes_ its output pipe. So, `tsp` just encounters a quite standard end-of-file. Can't do anything more, sorry...

I don't ask you for a solution. I'm only commenting it. Take note that _without_ the `ts::tsp::InputExecutor` class, all other TSDuck tools work without any problem! So, I'm suggesting only to be aware of this behaviour. Perhaps other processes will have a similar problem using PIPES.

closed 2019-06-25 by lelegard