tsduck/tsduck #341: ecm message

michaeladams123, 2019-06-25:
hi,
i have issue with some ecm when received to stb  ecm contain message to stb to show messages like hashcode watermark on TV and i need to filter or block it please i need your kindly support.

lelegard, 2019-06-25:
Is this a joke ?

Are you asking how to circumvent a security mechanism ? To me ? Please learn to use LinkedIn.

TSDuck is a toolbox for digital TV engineers to do test, monitoring, integration, debug, lab or demo. It has never meant to be used to implement security hacking practices. I cannot prevent anyone to use it outside a legal framework but you should at least have the decency to avoid publicly asking such a question.

You will __never__ get help from TSDuck to hack TV systems.

michaeladams123, 2019-06-26:
so sorry for that,sure not for hacking just we need to remove ads messages on stb like popup 
so sorry again

rezna11, 2019-06-26:
Pupup ads in ECMs? Weird. I would think you want to remove the serial from being shown on screen which is part of CAS. I've seen that stuff in India in hotels... all channels had a number appear on screen every couple of minutes, always in different place. So I would guess you want to remove it so that you can encode the channel for illegal IPTV streaming.

lelegard, 2019-06-26:
ECM’s are designed to transport information which is essential to the security of the system and pure commercial ads are not part of this. 

Users discussing security hacking will be banned from this forum.

michaeladams123, 2019-06-26:
sorry , we have legal IPTV and telecommunication company, and we are used your very power full tool to monitor the stream,and just we need to remove the hashcode because the provider can't customize it for one customer

michaeladams123, 2019-06-26:
sure no security hacking and so sorry for misunderstand

closed 2019-06-26 by lelegard