tsduck/tsduck #343: Cosmetic? "craft --no-payload" don't reset PUSI and Adaptation Field

lars18th, 2019-06-25:
Hi @lelegard ,

When using `-P craft --no-payload` the resulting packets have a payload size of zero. However, PUSI and Adaptation Field (with payload) continues if the original packet has it.

This is the expected behaviour?
In my opinion that doesn't have sense, but I prefer to ask about it.

Thank you!

lars18th, 2019-06-25:
Hi,

I forget to mention that for sure I can use `-P craft --no-payload --clear-pusi`, but this doesn't resolve full problem as the Adaptation Field continues indicating that a payload exists (and that's an error). Futhermore, I assume that `--no-payload` implies to reset the PUSI mark and the Adaptation Field.

I'm right?

lelegard, 2019-06-25:
The plugin `craft` is a very low-level tool. It is useful to have a way to create invalid packets for testing the robustness of applications. TSDuck is mainly used for tests. So, you need to specifiy `--clear-pusi`.

>  Futhermore, I assume that --no-payload implies to reset the PUSI mark ...

I may need a way to explicitly clear or set the payload indicator when the payload size is empty, for the same reason. This is why `--payload-size 0` should crunch the payload without resetting the payload indicator. But `--no-payload` should does this _and_ reset the payload indicator. I will fix this.

>  ... and the Adaptation Field.

If there is no payload, you must have a 184-byte adaptation field.

lars18th, 2019-06-25:
> The plugin `craft` is a very low-level tool. It is useful to have a way to create invalid packets for testing the robustness of applications. TSDuck is mainly used for tests. So, you need to specifiy `--clear-pusi`.

OK.

> I may need a way to explicitly clear or set the payload indicator when the payload size is empty, for the same reason. This is why `--payload-size 0` should crunch the payload without resetting the [...]

Great!

> > ... and the Adaptation Field.
> 
> If there is no payload, you must have a 184-byte adaptation field.

... or an empty packet with PCR! :wink:

lelegard, 2019-06-25:
> or an empty packet with PCR! 

A packet is never empty, the combination of payload and adaptation field is always 184 bytes long.

The PCR is part of the adaptation field. There is no PCR without adaptation field.

lars18th, 2019-06-25:
> > or an empty packet with PCR!
> 
> A packet is never empty, the combination of payload and adaptation field is always 184 bytes long.
> The PCR is part of the adaptation field. There is no PCR without adaptation field.

Sure! My response only exposes the reason to "craft" packets with `--no-payload`.
Thank you for your comments. :smiley: 
Regards.

lelegard, 2019-06-25:
The respective behaviours of `--no-payload` and `--payload-size 0` have been fixed in commit 173e9870a9e1357ff1510332495fe3e797e47b48, both in the input plugin and the packet processing one.

closed 2019-06-25 by lelegard

lars18th, 2019-06-25:
> The respective behaviours of `--no-payload` and `--payload-size 0` have been fixed in commit [173e987](https://github.com/tsduck/tsduck/commit/173e9870a9e1357ff1510332495fe3e797e47b48), both in the input plugin and the packet processing one.

Thank you! I'll check it soon.