tsduck/tsduck #344: Can't find craft.so

DenisVounckx, 2019-06-26:
Installed on Ubuntu 18.04 the version tsduck_3.17-1189_amd64.deb.
Tried to do: tsp -I craft
 -> tsp: craft.so: cannot open shared object file: No such file or directory

Looking for:  locate tsplugin_craft -> nothing found

But other plugings seem to exist:
locate tsplugin_
/usr/bin/tsplugin_aes.so
/usr/bin/tsplugin_analyze.so
/usr/bin/tsplugin_bat.so
/usr/bin/tsplugin_bitrate_monitor.so
/usr/bin/tsplugin_boostpid.so
/usr/bin/tsplugin_cat.so
/usr/bin/tsplugin_clear.so
/usr/bin/tsplugin_continuity.so
/usr/bin/tsplugin_count.so
/usr/bin/tsplugin_datainject.so
/usr/bin/tsplugin_decap.so
/usr/bin/tsplugin_dektec.so
/usr/bin/tsplugin_descrambler.so
/usr/bin/tsplugin_drop.so
/usr/bin/tsplugin_duplicate.so
/usr/bin/tsplugin_dvb.so
/usr/bin/tsplugin_eit.so
/usr/bin/tsplugin_encap.so
/usr/bin/tsplugin_file.so
/usr/bin/tsplugin_filter.so
/usr/bin/tsplugin_fork.so
/usr/bin/tsplugin_hides.so
/usr/bin/tsplugin_history.so
/usr/bin/tsplugin_hls.so
/usr/bin/tsplugin_http.so
/usr/bin/tsplugin_inject.so
/usr/bin/tsplugin_ip.so
/usr/bin/tsplugin_limit.so
/usr/bin/tsplugin_merge.so
/usr/bin/tsplugin_mpe.so
/usr/bin/tsplugin_mpeinject.so
/usr/bin/tsplugin_mux.so
/usr/bin/tsplugin_nit.so
/usr/bin/tsplugin_nitscan.so
/usr/bin/tsplugin_null.so
/usr/bin/tsplugin_pat.so
/usr/bin/tsplugin_pattern.so
/usr/bin/tsplugin_pcradjust.so
/usr/bin/tsplugin_pcrbitrate.so
/usr/bin/tsplugin_pcrextract.so
/usr/bin/tsplugin_pcrverify.so
/usr/bin/tsplugin_pes.so
/usr/bin/tsplugin_play.so
/usr/bin/tsplugin_pmt.so
/usr/bin/tsplugin_psi.so
/usr/bin/tsplugin_reduce.so
/usr/bin/tsplugin_regulate.so
/usr/bin/tsplugin_remap.so
/usr/bin/tsplugin_rmorphan.so
/usr/bin/tsplugin_rmsplice.so
/usr/bin/tsplugin_scrambler.so
/usr/bin/tsplugin_sdt.so
/usr/bin/tsplugin_sections.so
/usr/bin/tsplugin_sifilter.so
/usr/bin/tsplugin_skip.so
/usr/bin/tsplugin_slice.so
/usr/bin/tsplugin_spliceinject.so
/usr/bin/tsplugin_stuffanalyze.so
/usr/bin/tsplugin_svremove.so
/usr/bin/tsplugin_svrename.so
/usr/bin/tsplugin_t2mi.so
/usr/bin/tsplugin_tables.so
/usr/bin/tsplugin_teletext.so
/usr/bin/tsplugin_time.so
/usr/bin/tsplugin_timeref.so
/usr/bin/tsplugin_tsrename.so
/usr/bin/tsplugin_until.so
/usr/bin/tsplugin_zap.so

lelegard, 2019-06-26:
Hi @DenisVounckx,

The plugin `craft` is new. It is included in version 3.18, as indicated in the [release notes](https://tsduck.io/download/changelog/). So, it is normal you cannot find it in version 3.17.

Version 3.18 is still under development but stable. You can easily rebuild the `.deb` installer from the source repo, as [described here](https://tsduck.io/doxy/building.html). The process is straightforward: install the pre-requisite building tools, clone the repo, build.

On Ubuntu 18.04, this is as simple as:
~~~
apt install git g++ dpkg-dev doxygen dos2unix graphviz curl pcscd libpcsclite-dev libcurl4 libcurl4-openssl-dev
git clone https://github.com/tsduck/tsduck.git
cd tsduck
make deb
~~~
And you collect the `.deb` in the `installers` subdirectory.

DenisVounckx, 2019-06-26:
Hi @

Thanks for the rapid response.
Trying to build from source I have 2 errors:
```
[CXX] tsPacketizer.cpp
In file included from /usr/include/string.h:494:0,
                 from /usr/include/c++/4.9/cstring:42,
                 from tsPlatform.h:850,
                 from tsMPEG.h:37,
                 from tsPacketizer.h:36,
                 from tsPacketizer.cpp:34:
In function 'void* memset(void*, int, size_t)',
    inlined from 'bool ts::Packetizer::getNextPacket(ts::TSPacket&)' at tsPacketizer.cpp:207:44:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:67:32: error: call to '__warn_memset_zero_len' declared with attribute warning: memset used with constant zero length parameter; this could be due to transposed parameters [-Werror]
       __warn_memset_zero_len ();
                                ^
cc1plus: all warnings being treated as errors
../../Makefile.common:903: recipe for target 'release-x86_64/tsPacketizer.o' failed
```
To get pasted I disabled the Werror flag.
But then during linking I got following error:
```
  [LD] release-x86_64/tsanalyze
/home/dvo/tsduck/src/libtsduck/release-x86_64/tsduck.so: undefined reference to `__warn_memset_zero_len'
collect2: error: ld returned 1 exit status
../../Makefile.common:910: recipe for target 'release-x86_64/tsanalyze' failed
make[2]: *** [release-x86_64/tsanalyze] Error 1
```

lelegard, 2019-06-26:
What version of GCC do you have? Did you install some non-standard version of the compiler or the library?

It used to work on Ubuntu 18.04 with the "standard" GCC (I can't remember which version it was and now I use Ubuntu 19.04 with GCC 8.3.0). The code which triggers the error has not changed for years.

I ask the question about the compiler version because this really looks like a compiler error. The code is:
~~~
    // Fill the packet payload
    while (remain_in_packet > 0) {
        ....
            if (do_stuffing) {
                break;
            }
        ....
    }

    // Do packet stuffing if necessary.
    ::memset(data, 0xFF, remain_in_packet);  <--- error reported here
~~~

The error complains about `warning: memset used with constant zero length parameter`.

Here, the length parameter is `remain_in_packet`. The previous `while (remain_in_packet > 0)` may give the impression that `remain_in_packet` is always zero at the end of the loop. But, in fact, the `break` on `do_stuffing` proves the contrary.

So, this really looks like a compiler pretending to be very clever but failing miserably. A buggy compiler.

DenisVounckx, 2019-06-27:
Hi Thierry,

Trying with the default gcc on Ubuntu 18.04 (which is now gcc version 7.4.0) I received following messages:

```
#error "GCC versions 6 and 7 are broken and fail to properly handle some template substitutions, use another version of GCC, see https://tsduck.io/doxy/building.html#reqraspbian"
```
So I tried to install gcc 4.9 via "apt install gcc-4.9 g++-4.9" but didn't work so fall back to the following [procedure](https://askubuntu.com/questions/1036108/install-gcc-4-9-at-ubuntu-18-04)

This gave me "gcc version 4.9.3".

And indeed with compiler bug around memset. As a quick workaround changed the line to:

```
if (remain_in_packet > 0) ::memset (data, 0xFF, remain_in_packet);
```
And then it compiles. Will try later today with gcc 4.9.4

Thanks a lot for the help !

lelegard, 2019-06-27:
Glad to hear that it works now.

Just out of curiosity, which compiler did you use when you got the "memset" error?
I have never seen this and I am curious to understand where does it come from.

lelegard, 2019-06-27:
Glad to hear that it works now.

Just out of curiosity, which compiler did you use when you got the "memset" error?
I have never seen this and I am curious to understand where does it come from.

On Ubuntu 18.04, you can also use GCC 8, see https://packages.ubuntu.com/bionic/gcc-8. I use GCC 8 on Ubuntu 19.04 and GCC 9 on Fedora 30, they are all fine.

DenisVounckx, 2019-06-27:
Hi Thierry,

When the memset error appeared, I was using gcc version 4.9.3 (Ubuntu 4.9.3-13ubuntu2).
Thanks for the gcc 8 tip.

lelegard, 2019-06-27:
Hi @DenisVounckx,

For the record, the compilation is fine for me with GCC 4.9.3 (Raspbian 4.9.3-14) on my Raspberry Pi. This is not the same compiler because this is an ARM back-end. But the type of error you got is typically detected by the front-end. So, I would have expected to get the same error on ARM. For instance, GCC 6.x and 7.x have front-end bugs in the template substitutions and the same errors appear on Intel and ARM. So, your GCC 4.9.3 seems a weird one.

closed 2019-07-31 by lelegard