tsduck/tsduck #345: Request: NULL (packet processing) plugin

lars18th, 2019-06-26:
Hi @lelegard ,

I hope you don't think this is a stupid request:

- Can you add a new the dumb plugin called `null` (processing) or `pass`?

The complexity of the build toolchain of the TSDuck makes difficult to develop "new" plugins. So, if you add a simple `null`/`pass` plugin that only passes every packet (without any modification, and without parameters), then will be more easy to us to implement/test new plugin ideas. With this "template" we don't need to create new files, modify Makefiles, etc. Only do the coding inside the template plugin.

I'm sure this will be trivial for your.
Do you agree?

Regards.

lelegard, 2019-06-26:
Honestly, I do not like the idea at all. 

In the programmer's guide, in the [Developing TSP plugin](https://tsduck.io/doxy/devplugin.html), you have this guidelines:

> __Developing plugins for the TSDuck project__
>
> To develop a new plugin named foo, follow these steps:
>
> - Create a source file named tsplugin_foo.cpp in the tsplugins subdirectory.
> - On Linux or macOS systems, this new source file will be automatically recognized by the Makefile and the new plugin will be built.
> - On Windows systems, create a project file named tsplugin_foo.vcxproj in the msvc subdirectory. The fastest way is to copy another plugin's project file. Then, edit the file (this is an XML file) and replace the names of the project and source file. Finally, under Visual C++, add the new project in the solution (File / Add / Existing Project).

Otherwise, if you develop an independent plugin, you can start from the template in `sample/sample-plugin`.

This isn't that complicated.

lars18th, 2019-06-27:
> > To develop a new plugin named foo, follow these steps:
> > 
> > * Create a source file named tsplugin_foo.cpp in the tsplugins subdirectory.

That's just all I need.
Thank you!

closed 2019-06-27 by lars18th