tsduck/tsduck #346: Suggestion: delayer plugin

lars18th, 2019-06-26:
Hi @lelegard ,

I have an idea of an useful new plugin: the `delayer` plugin.

The concept is quite simple (but not trivial to implement): you select a pid and _all_ packets with this pid are converted to null and stored in an (small) internal buffer. The packets are then reinjected inside null packets after N seconds (you can use Time or Bytes as the delay value). Obviosly as the plugin uses memory the delay will be very small (around 10 sec. max depending on the memory and the bitrate).

What is your opinion about this idea?

lelegard, 2019-06-26:
Could be done, as many other ideas. There are just too many things to do these days. Remember that TSDuck is a single-person project and on spare time only.

lars18th, 2019-06-27:
OK.

Thank you! Maybe someday in the future. It's just an idea.

Regards.

lelegard, 2026-04-23:
Plugin `timeshift` does the job.