tsduck/tsduck #349: How to? Remap in interval

lars18th, 2019-06-29:
Hi,

Please, comment ideas about how to achieve this:

- I need to use the plugin `remap` to apply the remap function for _several_ intervals. Example:
 `-P remap 101=201 --interval 0-12345 -P remap 102=201 --interval 12346-20000 -P remap 103=201 --interval 20001-30000 ...`

I'm thinking on doing it using labels and the plugin `count`. However, the plugin `count` doesn't have the option `--set-label`; nor the plugin `remap` has the option `--interval`.

So before doing anything... Any idea about the most simple (and effective) way to do it?
Thank you for your comments.

lelegard, 2019-07-04:
Added option `--interval` to plugin `filter`. With the packet labelling mechanism, you can do anything on a specified list of intervals.

closed 2019-07-31 by lelegard

lars18th, 2019-08-27:
> Added option `--interval` to plugin `filter`. With the packet labelling mechanism, you can do anything on a specified list of intervals.

Thank you @lelegard ,

This enhancement is very powerful. :+1: 
However, I suggest to add some trigger (using labels) to reset the counter.

To be tranparent: The objective is to have a method using TSDuck to select (remapping) different video streams from the same source (at GOP boundaries). The final goal is a pseudo-statmux using different versions of the same video with different quality/size (instead of controlling the encoder to a target size/quality). Any idea about it?

Regards.