tsduck/tsduck #350: Question: VLC "underrun" when playing merged stream

felipewd, 2019-06-30:
Hi all,

I'm trying to understand a failure of mine when trying to use the merge plugin.

As an example, I'm trying to do the following:

- Read an input (actual is a network multicast, in this example is a file);
- Remove the 1st service and replace it with null packets;
- Merged it with _the same input file_;
- Play the output.

Such as:

`tsp --max-input-packets 16  -I file ./input.ts -P regulate -p 16 --wait-min 20 --pcr-synchronous -P svremove -e -s 0x1 -P merge "tsp --max-input-packets 16 -I file ./input.ts -P regulate -p 16 --wait-min 20 --pcr-synchronous" -P regulate -p 16 --wait-min 20 --pcr-synchronous -O play`

I'm also trying to control/regulate the input/processing/output all the way, because in my actual use case the Mux is _very_ sensitive.

The problem is: After around ~20-30 seconds VLC starts to output:

```
auhal warning: underrun of 4096 bytes
main warning: playback too late (65730): up-sampling
auhal warning: underrun of 4096 bytes
main warning: buffer too late (-71214 us): dropped
```

After a while of these...the sound stops completely.

Doesn anyone has any idea what the error might be? The real use-case for me is reading a network address and trying to replace it with this previously recorded-file...but the error is the same, and this is way easier to test.

Thanks!

lars18th, 2019-07-01:
Hi @felipewd ,

I see similar problems when muxing with the `merge` plugin. My solution is to use a different approach: If you read my different comments (issues and PR) you will be able to discover this other way of making the muxing process:

- For a correct muxing you need to prepare the _main_ TS and the _merged_ TS. The *requirement* is quite simple: both need to be CBR and with the same bitrate.
- Using current tools from TSDuck you can do it _increasing_ the bitrate of both streams with NULL stuff.
- Then you can mux them using the "one-to-one interleave mode" (not directly supported). For a realtime stream at time is not possible, however I feel @lelegard will implement it for the tswitch at some point. For files, you can use the `mux` plugin.
- After the muxing you have one TS with double the CBR of the input streams. Then you can eliminate the NULL stuff with the `reduce` plugin and recalculate the PCR timestamps with the `pcradjust` plugin. The only problem is how to achieve the correct _target_ bitrate (I'm discussing about this here: https://github.com/tsduck/tsduck/issues/322#issuecomment-505502378).

For me this the _only_ valid solution to do a correct muxing with the current TSDuck implementation.

PD: If you want to try, for realtime, test it with the `merge` plugin instead of the `mux` one.

I hope it helps.
Regards.

felipewd, 2019-07-01:
Hi @lars18th,

Thanks for the help!

One thing I don't understand, though:

- The file in question (test above) is a SPTS, with service `0x1`, with CBR (800Kbps);
- Since I'm removing it and replacing it with `null` packets, there's enough space to replace the content and to maintain the bitrate;
- As I'm replacing the removed it service with itself (merging the same file input read the same way as the main input), I can see no reason there should be a problem with the recomputed PCRs.

lars18th, 2019-07-02:
Hi @felipewd ,

> Thanks for the help!

:+1: 

> One thing I don't understand, though:
> * The file in question (test above) is a SPTS, with service `0x1`, with CBR (800Kbps);
> * Since I'm removing it and replacing it with `null` packets, there's [...]

Well. Some comments:

1. In my experience the `merge` plugin uses the PCRs.
2. And when you use the `regulate` plugin with the `--pcr-synchronous` you're using the PCRs too.
3. So, in this case you need to grant *two* requirements:
 -- Both streams (main and merged) requires to have _valid_ PCR timestamps.
 -- And both streams require to have known (and correct) bitrates.

So, in fact quite complex to achieve "perfect" (in the sense of _valid_) results. I'm not questioning the developer's competence, but I prefer another approach for a more _robust_ muxing than the `merge` plugin:

- What I'm doing at time is to "interleave" the main and the merged stream. You can do it if _both_ are CBR _and_ the bitrate is the same (or at least a multiple). See this "simple" example:
```
tsp -a 1/3 -I [whatever] ... \
 -P mux --set-label 1 --no-pid-conflict-check --no-continuity-update -i 4 file-to-merge.ts \
 ... \
 -P reduce XXX YYY \
 -P pcradjust --bitrate ZZZZZZ --ignore-dts --ignore-pts \
 -O [whatever]
```

Let me describe a few things about this example:
1. The first `-a 1/3` creates the space for the muxed TS. In this case the bitrate of the muxed TS be a _quarter_ (1/3 generates 1 null space every 3 of the input, so the bitrate increases in a quarter).
2. The `mux` pluing simply interleaves 1 packet of the muxed TS with 3 of the main TS. So if the bitrate is CBR and a quarter of the main TS then you're muxing "perfectly" (in the sense of _valid_).
3. After that you only need to merge or change the PSI/SI tables (the `...` zone).
4. And ajust the output bitrate... In this case you can use the `reduce` plugin to _remove_ the unused nulls (at time you need to calculate manually the values, waiting for https://github.com/tsduck/tsduck/issues/322#issuecomment-505502378) and use the `pcradjust` to recompute the PCR timestamps (the output bitrate requires to be manually forced, as the `reduce` plugin doesn't change the value. So be aware of the ZZZZZZ value).

I hope you want to check this other approach.
Regards.