tsduck/tsduck #351: Use diffent debug debug level for plugins

lars18th, 2019-07-01:
Hi,

I have a very stupid question, but I don't see the response:
- I need to enable debug _only_ for one plugin in a complex chain. Example:
`tsp -I file file1.ts ... -P filter -d4 ... -O drop`

What I want is to enable DEBUG only for the `filter` plugin, instead of a GLOBAL `tsp -d4`.
Any idea?

lelegard, 2019-07-02:
The verbosity level applies to a specific `ts::Report` object. Inside `tsp`, there is one centralized logger running in a specific thread with low priority. All threads and plugins send their logs there. So, the debug level applies to all.

Since debug is by definition used to debug, this is more a developer issue than a user issue. I suggest to grep for the plugin you need. E.g.
~~~
tsp -d4 ... 2>&1 | grep 'Debug.*: XXX:'
~~~
Where `XXX` is the name of the plugin to track.

lars18th, 2019-07-02:
> I suggest to grep for the plugin you need. E.g.
> ```
> tsp -d4 ... 2>&1 | grep 'Debug.*: XXX:'
> ```
> Where `XXX` is the name of the plugin to track.

Just what I need! :+1: 
I hadn't noticed that the name of the plugin is written on each log line.
Thank you!

closed 2019-07-02 by lars18th