tsduck/tsduck #352: Request: pcradjust plugin --add-empty

lars18th, 2019-07-01:
Hi @lelegard ,

The mux interleaved mode works very well (when using CBR inputs with the same bitrate). However, I have just one request about the plugin `pcradjust`:

- When making some modifications to the TS, it's possible that the distance of two consecutive PCR marks be greater than the required value of 40ms. Then I request to include the option `--add-empty [interval]` in order to include (reusing NULL stuff) an empty packet in the PCR pid with no-payload and only the PCR mark. The value `[interval]` is then the maximum time between marks.

Of course I don't request the removal of PCR marks when the distance is less than a value (although it could also be a good option, but it isn't necessary). I only request for this useful option.

Do you think it makes sense to implement it?
Regards.

lelegard, 2019-08-02:
Option `--min-ms-interval` added in commit 1e364430a31d1a97019a20879d1b165735a559be

lelegard, 2019-08-05:
Version 3.18 has been released. Please open a new issue in case of problem with it.

closed 2019-08-05 by lelegard