tsduck/tsduck #353: Possibility to move DektecDevice class out of private?

alevinsn, 2019-07-02:
Currently, the DektecDevice class header file, tsDektecDevice.h, is not made generally available and is located in the libtsduck\private directory.  It would be helpful to make this class part of the general TSDuck programming interface, however, since no equivalent functionality exists in TSDuck in terms of being able to programmatically query the existence of Dektec devices and learn about them prior to usage, likely either via the Dektec input or output plug-ins.  The DektecControl class doesn't really accomplish this, as it just outputs information to the console and doesn't provide a convenient way to programmatically perform queries.

The usage model is as follows:  at times, it would be nice to know if a Dektec device (input, output, etc) is available before attempting to use it.  While it is true that the Dektec input and output plug-ins' start() methods will fail if there is no Dektec device, they just return false to indicate this, and a return value of false can mean many different things.  It is necessary to check the log message reported to determine what caused the failure.  Plus, from a user interface perspective, it may be desirable not to show a choice of "Dektec" for input and/or output, and accomplishing this would be done most readily through the query-like functionality provided by the DektecDevice class.

While I know I can accomplish this by directly using the Dektec APIs, I think it would be convenient and preferable to be able to accomplish this entirely using TSDuck APIs.

alevinsn, 2019-07-02:
I see now that it isn't so simple, since the DektecDevice class makes direct use of DTAPI capabilities.  Nevertheless, it would still be helpful to provide something like DektecDevice but suitable for use without also needing to include DTAPI.h.

lelegard, 2019-07-02:
As you mention, the problem is not simple. The DTAPI is delivered in binary format as a `.lib` on Windows and `.o` on Linux. Since this is not a DLL / shareable object, it must be included by the linker in only one place. Currently, it is linked in `tsduck.dll` / `tsduck.so`. However, on Windows, it is impossible to export the DTAPI from `tsduck.dll` because the DTAPI header does not allow it. So, in a TSDuck application (at least on Windows), it is impossible to use the DTAPI outside `tsduck.dll`. This is why everything refering to the DTAPI is hidden inside the TSDuck library.

To make some features available, they must be clearly specified first. Then, they can be implemented in the TSDuck library. Currently, the only exported features are specified by the command `tsdektec` and the plugin `dektec`.

What are the features you would like to be exported from the TSDuck library, precisely?

alevinsn, 2019-07-02:
Considering a method like static UString GetDeviceDescription(const Dtapi::DtDeviceDesc&) in the DektecDevice interface, from the perspective of using this method, it could just as well be defined as static UString GetDeviceDescription(const DektecDevice *).  I'm thinking that it would be good to start small and only provide the functionality offered by the following two methods:  GetAllDevices() and GetDeviceDescription().  That would be sufficient for my purposes.  But, instead of adjusting the DektecDevice class, I propose the creation of a new class, something like AbstractDektecDevice.  It could be defined as something like:

class AbstractDektecDevice
{
    public:
        UString model;
        
        AbstractDektecDevice();

        static bool GetAllDevices(AbstractDektecDeviceVector&, Report& = CERR);
        static UString GetDeviceDescription(const AbstractDektecDevice&);
};

Internally, it would make use of the DektecDevice class, in a fashion similar to the DektecControl class.

Other functionality, such as GetAllPorts(), GetPortDescription(), and GetInterfaceDescription() would require a bit more work, since AbstractDektecDevice would need to offer the concept of a port in an abstract fashion, but I can accomplish what I'd like to do just using GetAllDevices() and GetDeviceDescription(), so I think that should be sufficient for now.

lelegard, 2019-07-03:
Added class `DektecDeviceInfo` in commit 66c588a8c18211f638f8a5f06afe6612349cbe7e. Please test.

It gives access to the information about Dektec devices and ports from the public API, but not to the devices themselves.

closed 2019-07-31 by lelegard