tsduck/tsduck #354: insert water mark logo in all programs

sachin0606, 2019-07-02:
Hi,

can we manipulate the TS programs, so can insert a watermark logo (image) on each program by using TSDuck?

lelegard, 2019-07-02:
This is video manipulation, not transport stream manipulation.

This is a job for ffmpeg, not for TSDuck.

FFmpeg manipulates the audio and video contents. TSDuck manipulates the transport stream around the content. Several people use a combination of ffmpeg and TSDuck to build a system because the two toolsets have complementary features (contents vs. envelope) and a similar filter / plugin mindset.

There are many examples for similar logo insertion with ffmpeg. See ffmpeg documentations and ffmpeg forums.

sachin0606, 2019-07-02:
Thanks for the info. will look into the respective documentation.

closed 2019-07-31 by lelegard