tsduck/tsduck #355: Question: How to copy components from one PMT to another

lars18th, 2019-07-02:
Hi,

I don't know if this can be done, so I describe the problem:

- I want to "copy" some components from one PMT of one service, to the PMT of another PMT. All without touching the source PMT.

What's the best solution to do it? I'm testing with the `sections` plugin but I don't know how to use it for this objective.

Any idea?

lars18th, 2019-07-02:
Hi @lelegard ,

Perhaps a simple solution can be this:
- A plugin (existing or new) that can "move" a sub-stream from one PMT to another PMT, including all descriptors associated with this sub-stream.

What do you think? This has sense, or it's possible in another way?
I like to know your opinion.
Regards.

lelegard, 2019-07-02:
It is possible to write such a plugin, but it I have no time to do so.

The simplest way is to extract the two PMT in XML format, do the modification in the XML and reinject the modified one.

lars18th, 2019-07-03:
> It is possible to write such a plugin, but it I have no time to do so.

Don't worry! I don't expect that you implement anything that we request.

> The simplest way is to extract the two PMT in XML format, do the modification in the XML and reinject the modified one.

Thank you for this tip! I'll use it.

Regards.

closed 2019-07-31 by lelegard