tsduck/tsduck #356: TSDuck Win64-3.17 Portable: Bug

lars18th, 2019-07-03:
Hi @lelegard ,

I discovered a strange bug inside the Windows Portable version 64bit (I don't know about other Windows versions)

- Command line:
```
E:
cd E:\TSDuck
.\bin\tsp.exe -d4 -I file file.ts -P zap 100 -O play
```

And I get this:
```
E:\TSDuck>.\bin\tsp.exe -d4 -I file file.ts -P zap 100 -O play
tsp: Debug[4]: debug level set to 4
* tsp options:
  --add-input-stuffing: 0/0
  --bitrate: 0 b/s
  --bitrate-adjust-interval: 5,000 milliseconds
  --buffer-size-mb: 16,777,216 bytes
  --debug: 4
  --list-processors: 0
  --max-flushed-packets: 0
  --max-input-packets: 0
  --realtime: unknown
  --monitor: 0
  --verbose: 1
  Number of packet processors: 1
  Input plugin 1:
    Name: file
    Type: input
    Arg[0]: "file.ts"
  Packet processor plugin 1:
    Name: zap
    Type: packet processor
    Arg[0]: "100"
  Output plugin 1:
    Name: play
    Type: output
* Debug[4]: debug level set to 4
tsp: Debug: trying to load "E:\TSDuck\bin\tsplugin_file.dll"
tsp: Debug: trying to load "E:\TSDuck\bin\tsplugin_play.dll"
tsp: Debug: symbol tspNewInput not found in E:\TSDuck\bin\tsplugin_play.dll
tsp: Debug: symbol tspNewProcessor not found in E:\TSDuck\bin\tsplugin_play.dll
tsp: Debug: trying to load "E:\TSDuck\bin\tsplugin_zap.dll"
tsp: Debug: symbol tspNewInput not found in E:\TSDuck\bin\tsplugin_zap.dll
tsp: Debug: symbol tspNewOutput not found in E:\TSDuck\bin\tsplugin_zap.dll
tsp: Debug: realtime: yes, using --max-input-packets 1,000 --max-flushed-packets
 1,000
* Debug[4]: file: debug level set to 4
* Debug[4]: zap: debug level set to 4
* Debug[4]: play: debug level set to 4
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
```

However, it's very strange as I use three different versions:
- Release TSDuck-Win64-3.16-1110-Portable.zip
- Release TSDuck-Win64-3.17-1189-Portable.zip
- Beta TSDuck-Win64-3.18-1283-Portable.zip

And with version 3.16 I don't see the error. With version 3.18 I see the error always. But with the version 3.17 I see the error sometimes only! I unzip the differents versions over the same `E:\TSDuck` directory and I execute the same command. So changing from one version to other, I obtain different behaviours with the version 3.17.

Any idea?
It's very annoying to me.

lelegard, 2019-07-03:
What is the error? What is the bug? I only see debug messages.

lars18th, 2019-07-03:
> What is the error? What is the bug? I only see debug messages.

The error? Nothing is displayed. The process crashes!

The bug? I feel is this...
```
[...]
tsp: Debug: symbol tspNewInput not found in E:\TSDuck\bin\tsplugin_play.dll
tsp: Debug: symbol tspNewProcessor not found in E:\TSDuck\bin\tsplugin_play.dll
[...]
tsp: Debug: symbol tspNewInput not found in E:\TSDuck\bin\tsplugin_zap.dll
tsp: Debug: symbol tspNewOutput not found in E:\TSDuck\bin\tsplugin_zap.dll
[...]
```

I don't think that these messages are normal.

lelegard, 2019-07-03:
> The error? Nothing is displayed. The process crashes!

You didn't mention it. Could you be more specific?

> `tsp: Debug: symbol tspNewInput not found in E:\TSDuck\bin\tsplugin_play.dll`
> `tsp: Debug: symbol tspNewProcessor not found in E:\TSDuck\bin\tsplugin_play.dll`

These messages are normal and have been present from the beginning, in all versions. They are debug messages only. Basically, the messages above just say that `tsp` could not find the entry points for the input plugin and packet processor plugin in `tsplugin_play.dll`, which is not surprising since `play` is an output plugin only. You can notice that there is no such message for `tsplugin_file.dll` because `file` can be used as an input, packet processor or output plugin. Thus, the three entry points are found.

lars18th, 2019-07-03:
> These messages are normal and have been present from the beginning, in all versions. They are debug messages only. Basically, the messages above just say that `tsp` could not find the entry points [...]

Ok, sorry! I feel the bug is related to this.

> > The error? Nothing is displayed. The process crashes!
> 
> You didn't mention it. Could you be more specific?

Well, I just unzip the Portable version to the `E:\` drive, and I copy the file `file.ts` to the directory `E:\TSDuck`. Then when I execute the command... the VLC doesn't appear and the tsp process crash. That's the behaviour using the Beta 3.18. I can't say more, except that I'm running in Windows 7 64bit (nothing special, because I use it for years, and the TSDuck Portable was working well until now).

Can you reproduce the crash?

lars18th, 2019-07-03:
Hi @lelegard ,

I discovered that the problem is related with the _FILE_ as if I use one or another different, then the behaviour is different. I'll try to isolate the problem...

However, I see that's very confusing that the `tsp` process with the debug enabled doesn't print the cause of the exit. Futhermore, I feel the problem is (perhaps) with the opening of the file. I suggest (if you agree) to add a litte more debug in the file reading.

Regards.

lars18th, 2019-07-03:
Hi @lelegard ,

This is completly annoying to me:

- `tsp -d6 -I file file1.ts > ./test.ts`: 0 bytes of output
- `tsp -d6 -I file file2.ts > ./test.ts`: identical files `test.ts` and `file2`

When I compare the debug output of both files the last lines are different:
- file1.ts with troubles
```
[...]
* Debug[6]: debug level set to 6
tsp: Debug: trying to load "E:\TSDuck\bin\tsplugin_file.dll"
tsp: Debug: trying to load "E:\TSDuck\bin\tsplugin_file.dll"
tsp: Debug: realtime: no, using --max-input-packets 0 --max-flushed-packets 10,000
* Debug[6]: file: debug level set to 6
* Debug[6]: file: debug level set to 6
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
```
- file2.ts without troubles
```
[...]
* Debug[6]: debug level set to 6
tsp: Debug: trying to load "E:\TSDuck\bin\tsplugin_file.dll"
tsp: Debug: trying to load "E:\TSDuck\bin\tsplugin_file.dll"
tsp: Debug: realtime: no, using --max-input-packets 0 --max-flushed-packets 10,000
* Debug[6]: file: debug level set to 6
* Debug[6]: file: debug level set to 6
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 31,668,466 b/s
* Debug: file: input thread started
* Debug: file: output thread started
```

Any idea?

lars18th, 2019-07-03:
Hi @lelegard ,

Take note that the problem only appears with the Portable **BETA 3.18**, with the previous 3.17 & 3.16 I don't see this behaviour. However, I need to use the HIDES modulator with a parameter (`-P timeref --eit`) that only exists in the last version. And it fails with the `file1.ts`, so it seems that the TSP 3.18 Beta refuses to process this file (and no more debug about the cause).

Any idea?

lars18th, 2019-07-03:
Hi @lelegard ,

This workaround resolves the problem with this `file1.ts`:
- `tsp -b 31668449 -I null -P regulate -P mux file1.ts ... `

However, it's a very ugly solution. Any idea why the `tsp` process doesn't like to start the processing when using some files? Even using pipes like `cat file1.ts | tsp ...` I obtain the same behaviour.

lelegard, 2019-07-03:
If a specific file creates an issue, I need that file to reproduce and analyze the issue. Could you please post this TS file somewhere?

lars18th, 2019-07-05:
> If a specific file creates an issue, I need that file to reproduce and analyze the issue. Could you please post this TS file somewhere?

Hi @lelegard ,

I'll try to isolate the problem and describe how to reproduce the problem generating a TS file that has troubles. :+1:

lars18th, 2019-07-09:
Hi @lelegard ,

Please note this executed on Linux (now I'm not sure that the problem is only with the Windows version):
```
tsp -d4 -I file file.ts -P tables -a -O null
tsp: Debug[4]: debug level set to 4
* tsp options:
  --add-input-stuffing: 0/0
  --bitrate: 0 b/s
  --bitrate-adjust-interval: 5,000 milliseconds
  --buffer-size-mb: 16,777,216 bytes
  --debug: 4
  --list-processors: 0
  --max-flushed-packets: 0
  --max-input-packets: 0
  --realtime: unknown
  --monitor: 0
  --verbose: 1
  Number of packet processors: 1
  Input plugin 1:
    Name: file
    Type: input
    Arg[0]: "file.ts"
  Packet processor plugin 1:
    Name: tables
    Type: packet processor
    Arg[0]: "-a"
  Output plugin 1:
    Name: null
    Type: output
* Debug[4]: debug level set to 4
tsp: Debug: trying to load "/root/tsduck/src/tsplugins/debug-x86_64/tsplugin_file.so"
tsp: Debug: trying to load "/root/tsduck/src/tsplugins/debug-x86_64/tsplugin_null.so"
tsp: Debug: symbol tspNewOutput not found in /root/tsduck/src/tsplugins/debug-x86_64/tsplugin_null.so
tsp: Debug: symbol tspNewProcessor not found in /root/tsduck/src/tsplugins/debug-x86_64/tsplugin_null.so
tsp: plugin null has no output capability
tsp: Debug: trying to load "/root/tsduck/src/tsplugins/debug-x86_64/tsplugin_tables.so"
tsp: Debug: symbol tspNewInput not found in /root/tsduck/src/tsplugins/debug-x86_64/tsplugin_tables.so
tsp: Debug: symbol tspNewOutput not found in /root/tsduck/src/tsplugins/debug-x86_64/tsplugin_tables.so
tsp: Debug: realtime: no, using --max-input-packets 0 --max-flushed-packets 10,000
```

And *NO* more output, the process exits.
But...

```
tsp -d4 -I file file.ts -P tables -a -O file tmp.ts
tsp: Debug[4]: debug level set to 4
* tsp options:
  --add-input-stuffing: 0/0
  --bitrate: 0 b/s
  --bitrate-adjust-interval: 5,000 milliseconds
  --buffer-size-mb: 16,777,216 bytes
  --debug: 4
  --list-processors: 0
  --max-flushed-packets: 0
  --max-input-packets: 0
  --realtime: unknown
  --monitor: 0
  --verbose: 1
  Number of packet processors: 1
  Input plugin 1:
    Name: file
    Type: input
    Arg[0]: "file.ts"
  Packet processor plugin 1:
    Name: tables
    Type: packet processor
    Arg[0]: "-a"
  Output plugin 1:
    Name: file
    Type: output
    Arg[0]: "tmp.ts"
* Debug[4]: debug level set to 4
tsp: Debug: trying to load "/root/tsduck/src/tsplugins/debug-x86_64/tsplugin_file.so"
tsp: Debug: trying to load "/root/tsduck/src/tsplugins/debug-x86_64/tsplugin_file.so"
tsp: Debug: trying to load "/root/tsduck/src/tsplugins/debug-x86_64/tsplugin_tables.so"
tsp: Debug: symbol tspNewInput not found in /root/tsduck/src/tsplugins/debug-x86_64/tsplugin_tables.so
tsp: Debug: symbol tspNewOutput not found in /root/tsduck/src/tsplugins/debug-x86_64/tsplugin_tables.so
tsp: Debug: realtime: no, using --max-input-packets 0 --max-flushed-packets 10,000
* Debug[4]: file: debug level set to 4
* Debug[4]: tables: debug level set to 4
* Debug[4]: file: debug level set to 4
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 31,668,463 b/s
* Debug: file: creating file tmp.ts, fd=4, error_code=0
* Debug: file: input thread started
* Debug: tables: packet processing thread started

* Debug: file: output thread started
* PAT, TID 0 (0x00), PID 0 (0x0000)
  Section: 0 (last: 0), version: 8, size: 48 bytes
...
```

And in this case all goes well. Futhermore, a simple `tstables file.ts` works. Annoying?

I do some debug and I see that the in first case the process exits because the variable `_is_valid` is false. But at time it's impossible to know what module has set it to false. So I suggest to make a small change in the source to DEBUG what module has set it to false.

You agree?

lars18th, 2019-07-09:
Hi @lelegard ,

Don't care about the last message... it's fixed after a full *recompile* (in Linux without a `make clean` you can be in troubles). However, I need to apply the PR #362 . So, I recommend after merging this PR (or fixing the problem) to compile a new Windows pre-release version for testing it.

Regards.

lelegard, 2019-07-31:
Do you still have the problem with the same files using the latest version?

lelegard, 2019-08-05:
Version 3.18 has been released. Please open a new issue in case of problem with it and the same files. But please provide input files which reproduce the problem.

closed 2019-08-05 by lelegard