tsduck/tsduck #358: Question about multiple plugins changing PSI/SI tables

lars18th, 2019-07-05:
Hi @lelegard ,

I ask this because I don't know the response. Please don't think this issue is a criticism in any way:

- When we're using multiple times the same plugin, for example: `tsp ... -P svrename -n "ch1" 100 -P svrename -n "ch2" 200 ... -P svrename -n "ch9" 900 ... `, in this case the tables are readed 9 times, rebuilded 9 times and writed 9 times for each table in the input?

The reason to this question is because when I do some precessing using such plugins with _FILES_, the regular tables appears far of the start of the file. So I want know the reason. And I feel that in cases like the one described the resulting table appears with a delay of 9 times after the first incoming table. It's my assumption true?

Thank you for listening to us, even when we ask absurd questions.
Regards.

lelegard, 2019-08-01:
In order to modify a table, a plugin first needs to analyze it completely. The analysis and modification can only start at the last TS packet for this table. The replacement of the table starts at the last TS packet of the table. The previous TS packets for the first instance of the table are replaced with null packets.

So, if the table is very short and fits in one packet, it should be modified "in place". Otherwise, each replacement tends to shift it ahead in the stream.

For a generic stream processing, there is no alternative because you cannot go backward.

In the case of a file, I addressed this issue with a script. 

For instance, in some use cases, with SPTS files, there is a requirement that the PAT fits in the first TS packet, immediately followed by the PMT, before any other type of content. When the SPTS comes from a stream capture, even without table modification, the requirement is not fullfilled. So, since we work on files, we can process them (or part of them) several times. So, my script processes the beginning of the file (using plugin `until`), modifying and extracting the tables of interest in binary files. Then, the file is then processed with option `--add-start-stuffing` and the tables are reinserted.

Sample excerpt from my script:
~~~
tsp --bitrate $TS_BITRATE --add-start-stuffing $INIT_PACKETS --add-input-stuffing 1/20 \
    -I file "$INFILE" \
    -P filter --pid 0 --pid $PMT_PID --negate --stuffing \
    -P inject "$TMPFILE.pat.bin" --pid 0 --bitrate $PAT_BITRATE --stuffing \
    -P inject "$TMPFILE.pmt.bin" --pid $PMT_PID --bitrate $PMT_BITRATE --stuffing \
    ...
    -P filter --after-packets $INIT_PACKETS --pid 0x1FFF --negate \
    -O file "$OUTFILE"
~~~

lelegard, 2019-08-01:
To be generic, the various characteristics of the file are extracted first:
~~~
tsp -I file "$INFILE" \
    -P until --packets 10000 \
    -P analyze --normalized -o "$TMPFILE.norm.txt" \
    -O drop

# Get TS bitrate in b/s, service id, PAT and PMT PID and bitrate.
TS_BITRATE=$(grep -m 1 '^ts:' "$TMPFILE.norm.txt" | sed -e 's/.*:bitrate=//' -e 's/:.*//')
PAT_BITRATE=$(grep -m 1 "^pid:pid=0:" "$TMPFILE.norm.txt" | sed -e 's/.*:bitrate=//' -e 's/:.*//')
SERVICE_ID=$(grep -m 1 '^service:' "$TMPFILE.norm.txt" | sed -e 's/.*:id=//' -e 's/:.*//')
PMT_PID=$(grep -m 1 '^service:' "$TMPFILE.norm.txt" | sed -e 's/.*:pmtpid=//' -e 's/:.*//')
PMT_BITRATE=$(grep -m 1 "^pid:pid=$PMT_PID:" "$TMPFILE.norm.txt" | sed -e 's/.*:bitrate=//' -e 's/:.*//')
~~~

lars18th, 2019-08-27:
Hi @lelegard ,

Thanks for the information. Your experience is very extensive and valuable.
I will try to use it.

Regards.