tsduck/tsduck #363: Build fails using GCC 4.8 on Ubuntu 18.04

rubenochiavone, 2019-07-10:
Hi there,

Build fails using GCC 4.8 on Ubuntu 18.04 (ref 701830b3). To reproduce:

```
make -j8 CC=gcc-4.8 GCC=gcc-4.8 CXX=g++-4.8 AR=gcc-ar-4.8 NOPCSC=1 NOCURL=1 NODTAPI=1
```

Log

```
  [CXX] tsPluginOptions.cpp
  [CXX] tsPluginRepository.cpp
In file included from /usr/include/string.h:494:0,
                 from /usr/include/c++/4.8/cstring:42,
                 from tsPlatform.h:850,
                 from tsMPEG.h:37,
                 from tsPacketizer.h:36,
                 from tsPacketizer.cpp:34:
In function ‘void* memset(void*, int, size_t)’,
    inlined from ‘bool ts::Packetizer::getNextPacket(ts::TSPacket&)’ at tsPacketizer.cpp:207:44:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:67:32: error: call to ‘__warn_memset_zero_len’ declared with attribute warning: memset used with constant zero length parameter; this could be due to transposed parameters [-Werror]
       __warn_memset_zero_len ();
                                ^
cc1plus: all warnings being treated as errors
../../Makefile.common:903: recipe for target 'release-x86_64/tsPacketizer.o' failed
make[2]: *** [release-x86_64/tsPacketizer.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

Looking at the source of the build error it was simple to fix:

**src/libtsduck/tsPacketizer.cpp**

```diff
@@ -204,7 +204,9 @@ bool ts::Packetizer::getNextPacket(TSPacket& pkt)
     }
 
     // Do packet stuffing if necessary.
-    ::memset (data, 0xFF, remain_in_packet);
+    if (remain_in_packet) {
+        ::memset (data, 0xFF, remain_in_packet);
+    }
     return true;
 }
 
```

After that I could build and run `tsduck` binaries.

Btw, thank you for such great project.

lars18th, 2019-07-10:
Hi @rubenochiavone ,

At time, for Debian I use the GCC 4.9 compiler. However, if this simple change fixes the compilation with the GCC 4.8 version, then it will be desirable to accept it. As more compilers accept the source... :wink: 

Regards.

rubenochiavone, 2019-07-11:
> At time, for Debian I use the GCC 4.9 compiler.

I guess this the recommended version as per the docs https://tsduck.io/doxy/building.html#requbuntu.

> However, if this simple change fixes the compilation with the GCC 4.8 version, then it will be desirable to accept it. As more compilers accept the source...

Glad to know.

I sent a PR to fix it. Hope it helps.

BR

PR #365 (2019-07-28): Fix memset warning when bytes length is zero
Since 'all warnings being treated as errors' GCC 4.8 complains about memset with zero bytes length and compilation fails. Skipping memset when remain_in_packet is zero fixed the issue.

Fixes #363.
closed 2019-07-28 by lelegard