tsduck/tsduck #368: Change DVB Tuning While TSP is Running?

SS-FF, 2019-07-12:
The tsscan tool in its current form is too limited for real-world channel scanning because it can only do NIT based scanning (except for terrestrial).

That's why I guess that I need to do channel scanning via TSP instead. For this purpose, I'm wondering whether it's possible to change tuning even while TSP is running in order to avoid starting a new process for each frequency to scan?

SS-FF, 2019-07-30:
Spawning the tsp process a hundred times for a detailed satellite scan using transponders lists instead of NIT is working fine actually.

I'd still be interested in your opinion about whether this could be done with a single execution of TSP where the tuning is changed while TSP is running.

Do you think it could be achieved with reasonable effort at all?

lars18th, 2019-07-30:
Hi @SS-FF ,

I've made some processing using *two* TSP process, and it works well. So I suggest to try this:
- Use the `tsswitch` to manage two inputs, and connect the output of it with a pipe with your regular TSP process. Then use two instances of the TSP for tunning and change the tuner parameters restarting the processes and selecting input 1 or 2.

I hope it helps.
Regards.

lelegard, 2019-07-30:
The `tsp` model is based on the independence between plugins. When you change the tuning, you change the input. So, this is a job for the input plugin. But the analysis is the job of a packet processor plugin. What you want to do does not fit well into the `tsp` architecture. This is why the specialized `tsscan` utility was developed.

Generally speaking, this is a common error to want one single instance of `tsp` to do everything in all cases. There are cases where this is fine, and others where this is not.

TSDuck is a toolbox. In itself, it does nothing. The engineer using TSDuck does everything. TSDuck was designed with scripting in mind. Complex cross-TS operations are done using scripts. Whether you use bash, PowerShell, Python or whatever you want as a scripting language, TSDuck gives a technical access to the TS and the engineer (you) has to develop his own logic using scripting.

Some utilities or plugins have a `--normalized` option to generate a form of output which can be deterministicly and easily parsed by scripts using basic tools such as `grep` and  `sed`.

In the TSDuck user's guide, look at the following examples:

- 5.2.13 Analyzing the bitrate of all services in a network: A shell script to build a synthetic report of the bitrates of all services in all TS in a network.
- 5.2.12 Monitoring the stuffing rate of all transponders in a network: A shell script to give a synthetic view of the stuffing bitrate of all TS in a network over time, building graphs using Excel.
- 5.2.11 Performing the global analysis of a network: Analyzing a complete DVB-T network using a makefile instead of a script. Yes, `make` can be a form of scripting language.
- 5.2.8 Scanning all services by CAS operator: A shell script to build a list of all services in a network for a given CAS operator.

Many of these examples demonstrate the power of the `--normalized` options.

lars18th, 2019-07-30:
> Generally speaking, this is a common error to want one single instance of `tsp` to do everything in all cases. There are cases where this is fine, and others where this is not.

This is something I've learned by using it. Now I agree with that statement. :+1: 

> TSDuck is a toolbox. In itself, it does nothing. The engineer using TSDuck does everything. TSDuck was designed with scripting in mind. Complex cross-TS operations are done using scripts.

Yes, we need to use scripts for controlling TS processes. That's a basic thing. However, IMHO I feel that some tasks can be simplified incorporating more options to some plugins.

Regards.

SS-FF, 2019-07-30:
> The `tsp` model is based on the independence between plugins. When you change the tuning, you change the input. So, this is a job for the input plugin. But the analysis is the job of a packet [...]

Thanks for the reply. Maybe I should have added that I'm using a custom implementation for mpegts parsing in most cases, so it's TSP often just doing dvbinput >> ip output. 

So I totally understand when you say that changing tuning while TSP is running does not fit into the architecture.

Question answered, thanks!

closed 2019-07-30 by SS-FF