tsduck/tsduck #369: Graceful Shutdown of TSP.exe on Windows

SS-FF, 2019-07-12:
When running TSP with the DVB input plugin on Windows, some BDA drivers don't react well when the tsp process is terminated in a hard way, i.e. via task manager, taskkill or similar.

Those BDA filtergraphs are using kernel streaming which means that data is crossing the border between user mode and system/kernel. Ideally running filtergraphs should be stopped (and even better: cleaned up) before the TSP process ends.

This is no problem when running TSP in a console: You can press CTRL-C (or CTRL-BREAK if you had implemented it) which allows the app to do all the cleanup.

But that's impossible when the TSP.exe is launched as a child process without console. It's a much bigger problem than one would expect in the first place:

- It doesn't work to send/write the key combination CTRL-C to the StandardInput of the TSP.exe process because CTRL-C is a control code, not simply a key code (combination). The TSP.exe will receive CTRL-C as data input but not a control code
- Then there's the **GenerateConsoleCtrlEvent** Win32 API which is meant to do exactly what's required: Sending the CTRL-C  control to a process  
But for some obscure reason, this doesn't work when a parent process sends that to a child process (https://www.google.com/search?q=GenerateConsoleCtrlEvent+child+process)
- Then things are getting kind of pervert: To overcome the parent=>child process signaling problem, several applications have evolved for sending a CTRL-C signal to a process  
That means: You start a new process in order to end your child process....

I really don't want to do such kind of things. There must be a better way. But the only other idea I had was to have TSP.exe listen on STDINPUT for a certain sequence.

What do you think about it, do you have any other idea?

lelegard, 2019-07-30:
With the version 3.18 (not yet released, you need to build it), you can use the `cutoff` plugin. Send the string "exit" using UDP (`echo >/dev/udp/127.0.0.1/NNN exit` with NNN being the listening port number). This generates an end of stream and `tsp` gracefully terminates.

SS-FF, 2019-07-30:
That sound like a good option.
Meanwhile I've implemented shutdown via key command from STDIN. It's very similar to the way ffmpeg is handling this. I could submit a PR, but I'm not sure if that would match your architecture..?

lelegard, 2019-07-30:
> Meanwhile I've implemented shutdown via key command from STDIN.

Is this compatible with situations where stdin is a transport stream (the default when no input plugin is specified) ?

Is this portable ? Windows / Linux / macOS ?

Otherwise, the `cutoff` solution seems sufficient. Generally speaking, such a feature should be implemented as a plugin. When the plugin receives the "signal" - whatever it is - it simply returns `TSP_END` on the next packet and `tsp` terminates.

SS-FF, 2019-08-01:
I'm quite familiar with ffmpeg as I've done some customizations and development there, so I've roughly followed the pattern that is used there.
Portability would be 'yes'.
Of course it cannot work when streaming via STDIN.

But I'd see it like this:
- Just for receiving a quit command it's probably not worth to take the effort and introduce this into the code base
- But if you would see some value in introducing an "interactive command mode" in general allowing for other additional commands (similar to ffmpeg), then it would be the way to go to

lelegard, 2019-08-02:
> Of course it cannot work when streaming via STDIN.

So, let's not do it this way.

> Just for receiving a quit command it's probably not worth to take the effort and introduce this into the code base

It is already part of the plugin `cutoff` anyway.

> But if you would see some value in introducing an "interactive command mode" in general allowing for other additional commands (similar to ffmpeg), then it would be the way to go to

What kind of interactive commands would you see? I use ffmpeg from time to time for transcoding operations. But I am not as familiar with it as you are.

SS-FF, 2019-08-02:
You can increase or decrease log level via '+' and '-' keys.
You can send commands to filters for changing parameters.
You can quit.
You can change debug mode.
You can dump packets.

But that's ffmpeg and this is tsDuck. tsDuck is a great achievement that I do respect that a lot. 
I got my own case solved anyway - the reason I asked about this was just driven by the idea of giving something in return - in case you would have replied something like that you've already planned to do some kind of interactive control mode.. ;-)

So, nevermind and thanks for publishing that library!

closed 2019-08-02 by SS-FF

lelegard, 2019-08-02:
OK. So let's stay with what is present today using the `cutoff` plugin.

- Exit command.
- Play with labels to affect subsequent plugins behaviour.