tsduck/tsduck #370: How to smooth PSI/SI tables ?

lars18th, 2019-07-18:
Hi @lelegard ,

I need your help and experience. I'm sure you know the most simple and effective way to accomplish this task. First let me to explain the problem:

- I'm using the approach for muxing based on the one-to-one interleaving and a final stage of the "reduce" plugin. More or less all works without troubles, except one thing.

- The repetition interval of the PSI/SI tables is greater than the values expected by the standards. In the entire processing chain I always use the parameters to "replace" the original tables. But after the muxing process and the "reduce" procedure in a lot of cases the repetition interval is a little more than the maximum time (and sometimes it's not really little).

Then I'm searching for a solution to this problem based on this simple idea:
```
tsp ... -P reduce AA BBB \
 | tsp -a 1/50 -P smooth-tables -P pcradjust --bitrate XXX --ignore-dts --ignore-pts -O ...
```

As you can imagine, the first tsp process is doing a large processing (it contains a dozens of plugins). And the second is for targeting the conformance of the TS. First it adds a small number of null stuffing, re-insert the tables in the correct place (I called it "smooth-tables"), and finally it recalculates all PCR marks with the final bitrate.

I think that's the best way to generate a compliant MPTS. However, I still need to know which is the best option to carry out the task of "smoothing the tables".

Please, can you tell me what might be the best method to do this?
I really appreciate your help, and the great work you're doing with this project.

Regards.

lars18th, 2019-07-29:
Hi @lelegard ,

Please can you provide _some idea_?
I suppose, as usual, there will be different alternatives. And for the moment, any solution (even if it's cumbersome) that works with the current implementation would be enough for me.

Thank you very much for your help!

lelegard, 2019-07-29:
Maybe you could extract the tables in a file (one file per table) and reinject it with a different bitrate or packet interval. Make sure to start with a clean area for those files and use option `--rewrite-binary` with plugin `tables` and option `--poll-files` with plugin `inject`.

The script would be something like (untested):
~~~
rm -rf /some/temp/directory
mkdir -p /some/temp/directory

tsp ....
    -P tables --pid XXX --tid YYY --binary-output /some/temp/directory/table-XXX-YYY.bin --rewrite-binary \
    -P filter --pid XXX --negate --stuffing \
    ...
    -P inject --pid XXX  --bitrate ZZZ /some/temp/directory/table-XXX-YYY.bin --poll-files --stuffing \
    ...
~~~

lars18th, 2019-07-29:
Thank you! I'll try this first.

Regards.

lars18th, 2019-07-30:
Hi @lelegard ,

Testing it...

> Make sure to start with a clean area for those files and use option `--rewrite-binary` with plugin `tables` and option `--poll-files` with plugin `inject`.

What do you think? It has sense to add some option like `--binary-delete` to delete the binary files at start and end points? This will ensure that no files are found from a previous execution. It has relevance when the process kills or something similar.

lars18th, 2019-07-30:
Hi @lelegard ,

It seems to work. Thank you! :+1: 

Now I suggest to update the documentation with a new example: _How to smooth PSI/SI tables_. This makes sense because after a major modification of the TS it may be necessary to smooth the tables to meet the standards.

Here an example:
```
tsp ... \
    ... \
 -P tables  -p  0 --binary-output table.pat --rewrite-binary \
 -P tables  -p  1 --binary-output table.cat --rewrite-binary \
 -P tables  -p 16 --binary-output table.nit --rewrite-binary \
 -P tables  -p 17 --binary-output table.sdt --rewrite-binary \
 -P filter  -p  0 -n -s \
 -P filter  -p  1 -n -s \
 -P filter  -p 16 -n -s \
 -P filter  -p 17 -n -s \
 -P inject  -p  0    -s --binary table.pat=400  \
 -P inject  -p  1    -s --binary table.cat=400  \
 -P inject  -p 16    -s --binary table.nit=3000 \
 -P inject  -p 17    -s --binary table.sdt=1000 \
 -O ...
```

Some comments:
- As you can observe, it's preferable to use the repetition interval to meet the specifications.
- This processing have sense at the end of the processing.
- In any case I suggest to add to the `tables` plugin then option `--binary-clean` or `binary-rewrite`.
- I don't know if this processing can work with EIT... any idea?
- It can be also useful to apply it to the PMT tables of each service.

Regards.

lars18th, 2019-08-30:
Hi @lelegard ,

I hope you agree to add this example (_How to smooth PSI/SI tables_) to the list of examples in the Section 5.2 of the documentation.

Here one improvement over the last example:

```
tsp -a 1/1 ... \
    ... \
 -P tables  -p  0 --binary-output table.pat --rewrite-binary \
 -P tables  -p  1 --binary-output table.cat --rewrite-binary \
 -P tables  -p 16 --binary-output table.nit --rewrite-binary \
 -P tables  -p 17 --binary-output table.sdt --rewrite-binary \
 -P filter  -p  0 -n -s \
 -P filter  -p  1 -n -s \
 -P filter  -p 16 -n -s \
 -P filter  -p 17 -n -s \
 -P inject  -p  0    -s --binary table.pat=400  \
 -P inject  -p  1    -s --binary table.cat=400  \
 -P inject  -p 16    -s --binary table.nit=3000 \
 -P inject  -p 17    -s --binary table.sdt=1000 \
 -P reduce  1 2 \
 -O ...
```

Using this technique at the end of the processing the repetition intervals are nearly perfect, and the structure of the output TS remains unchanged. However, I recommend to add some `-P pcradjust   --bitrate {TARGET-BITRATE} --ignore-dts --ignore-pts --min-ms-interval 40` just after the `reduce` plugin to clean the PCR timestamps.

Regards.