tsduck/tsduck #372: tsanalyze with UDP multicast input

stoyanovgeorge, 2019-07-18:
I am trying to use `tsanalyze` with UDP multicast input as a source. According to the manual, `tsduck` is only accepting file input but I  could use also a pipe from a live stream. The problem is that I don't really understand how I can do that. Is there an example of such use case from where I can refer to the exact syntax of the `[input-file]`?

yaneti, 2019-07-18:
Using a pipe would mean feeding ```tsanalyze``` the multicast on its standard input.  How you do that depends on your OS. 

Generally ```tsp``` is better suited syntax-wise for this usecase, but if you insist on ```tsanalyze``` here is
an example for Linux of 10 seconds of a UDP multicast group 224.24.24.24 on port 9999 that you are setup to be able to receive on eth0 (for socat) or via your default route (for multicat)

with ```socat```
```
# timeout --foreground 10 socat  UDP4-RECV:9999,ip-add-membership=224.24.24.24:eth0 - | tsanalyze
```

with ```multicat```
```
# timeout --foreground 10 multicat -X  -u -U @224.24.24.24:9999  /dev/null | tsanalyze
```

stoyanovgeorge, 2019-07-18:
Thanks a lot. I am also using Linux so I think this will suit me well. The reason why I want to use `tsanalyze` and not the `analyze` plugin is because I want to run `error-analysis` which I think is only present in `tsanalyze`. Otherwise, I need to first save the input multicast to a file and then run the `error-analysis`.

closed 2019-07-18 by stoyanovgeorge

rodrigodalri, 2019-12-26:
I try this command and my output is empty.
[output.txt](https://github.com/tsduck/tsduck/files/4002676/output.txt)

lelegard, 2019-12-31:
All options from `tsanalyze`, including `--error-analysis`, are also present in the plugin `analyze`. The code is common between the two.

What you want to do is this:
~~~
tsp -I ip ... -P until --seconds 10 -P analyze ... -O drop
~~~