tsduck/tsduck #374: `make` exists with status 0 even though compiling failed

masnagam, 2019-07-20:
I expected that `make` exists with status non-zero (2) if compiling failed. But `make` exits with status 0 actually.

I confirmed this issue on:

* macOS + clang-1001.0.46.4
* [Travis CI: osx environemnt](https://travis-ci.com/masnagam/tsduck-arib/jobs/217756291#L884)

I have not checked whether the same issue occurs on other platforms.

Normally, `make` exists with status 2 if compilation failed. I confirmed this behavior with the following simple Makefile on macOS:

```Makefile
# Makefile
foo.o: foo.cpp
	$(CXX) -o foo.o foo.cpp
```
```c++
// foo.cpp
x
```

I think that `make` should exit with status non-zero for safety if compiling failed. Because the current behavior causes the following issue:

```console
$ make -j $(nproc) NODTAPI=1 NOPCSC=1 test
  [REBUILD] libtsduck-files.props
  [REBUILD] libtsduck-filters.props
  [REBUILD] libtsduck-files.pri
  [DEP] tsAACDescriptor.cpp
  [CXX] tsAACDescriptor.cpp
tsAACDescriptor.cpp:1:1: error: unknown type name 'x'
x//----------------------------------------------------------------------------
^
In file included from tsAACDescriptor.cpp:34:
In file included from /Users/masnagam/workspace/tsduck/src/libtsduck/tsAACDescriptor.h:36:
In file included from /Users/masnagam/workspace/tsduck/src/libtsduck/tsAbstractDescriptor.h:36:
In file included from /Users/masnagam/workspace/tsduck/src/libtsduck/tsAbstractSignalization.h:36:
In file included from /Users/masnagam/workspace/tsduck/src/libtsduck/tsAbstractDefinedByStandards.h:36:
In file included from /Users/masnagam/workspace/tsduck/src/libtsduck/tsMPEG.h:37:
In file included from /Users/masnagam/workspace/tsduck/src/libtsduck/tsPlatform.h:781:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/cerrno:26:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__config:510:1: error: expected unqualified-id
namespace std {
^
2 errors generated.
make[2]: *** [release-x86_64/tsAACDescriptor.o] Error 1
source release-x86_64/setenv.sh && release-x86_64/utest

OK (390 tests, 8608 assertions)

source release-x86_64/setenv.sh && release-x86_64/utest_static

OK (387 tests, 8596 assertions)
```

`make test` runs utest even when compiling failed if utest already exists.

I checked Makefile files, but I couldn't figure out root causes like `-` before commands.

### Reproduction Steps

1. Run `make` and `make test`
2. Confirm that no build error and test failure occur
3. Modify a source file in order to cause compilation errors
4. Run `make` or `make test`
5. `make` exists with status 0

lelegard, 2019-07-29:
Fixed in commit 2f0a2770d43682d025149e5f4592dbc39e9c5a88

masnagam, 2019-07-30:
Confirmed that this issue has been fixed in the latest master.

Thank you for your effort!

closed 2019-07-30 by masnagam