tsduck/tsduck #375: How to keep send SCTE35 pagkage that add-pid  in PMT Table

edwardlai, 2019-07-22:
Input UDP (no SCTE35 pid)
Output UDP (have SCTE35 pid)

Command bellow 
tsp -I ip 227.7.7.7:7654 -P pmt --service 22 --add-programinfo-id 0x43554549 --add-pid 600/0x86 -P spliceinject --service 22 --files spliceload\splice*.xml  -O ip 226.6.6.6:4567

lars18th, 2019-07-24:
Hi @edwardlai ,

> But how to keep sent SCTE35 packet . otherwise TR101/290 will detect PID_error
> "Referred PID does not occur for more than 5 seconds, pkt 133084"

Try to use the optional parameter `--inject-interval` for the plugin `spliceinject`. Example:
`tsp ... -P spliceinject --inject-interval 500 --service 64 --files spliceload\splice*.xml ...`

Read the documentation at page 224 for the section "Spliceinject".
Regards.

edwardlai, 2019-07-29:
> Try to use the optional parameter `--inject-interval` for the plugin `spliceinject`. Example:
> `tsp ... -P spliceinject --inject-interval 500 --service 64 --files spliceload\splice*.xml ...`

Tried  Add --inject-interval 500 
Still get PID_error message after saved XML 5 seconds past .

Referred PID does not occur for more than 5 seconds, pkt 204558

Current Command 
tsp -I ip 227.7.7.7:7654 -P pmt --service 64 --add-programinfo-id 0x43554549 --add-pid 600/0x86 -P spliceinject --inject-interval 500 --service 64 --files spliceload\splice*.xml  -O ip 226.6.6.6:4567

lelegard, 2019-07-29:
> "Referred PID does not occur for more than 5 seconds, pkt 133084"

That is the way SCTE-35 works. The PID is referred in the PMT but it does not carry any information except a few packets before an event. If there is no event during hours, there won't be any packet in the SCTE-35 PID.

The requirement to have at least one packet one the SCTE-35 PID every 5 seconds is not compatible with SCTE-35.

And remember that SCTE-35 is an SCTE standard, not a DVB standard. If you want to implement SCTE-35 in a DVB network, you have to ignore the "referred PID" timing requirement.

You can still insert null events from time to time but this is not something that TSDuck does. And this is functionally useless anyway.

closed 2019-12-02 by lelegard