tsduck/tsduck #380: Feedback for hardware to use TSDuck on Production

cdavid14, 2019-07-25:
Hi everyone, i'm looking for an feedback about the hardware you guys using on production for your applications.

The most common use for tsduck in the region here is for ip remux, the input is an UDP that we need to remap to an UDP output, most times taking one UDP input with X programs and remap to multiple UDP outputs and my doubt is the criteries for do this, how much cpu/memory/gpu i need to each program, things like this.

I'm newbie on this world and i'm trying to get more information with the experts on the area, i'm an developer and my objective is contribute with this project as i can!

lelegard, 2019-07-28:
Hi @cdavid14 

The first thing I would like to make clear is that TSDuck has been designed for test, demo, debug, etc. It has not been designed with production in mind. That being said, TSDuck has proven to be robust but it should be clear that using it on production will be done at your own risk.

For production use, I recommend to use Linux, not Windows.

Taking one MPTS as UDP input and sending as many SPTS as there are services is trivial and uses very little resource. Any regular modern PC with a few GB of RAM is sufficient. No GPU is required.

The best benchmark is to try it yourself.

The typical command skeleton would be something like this (with 4 input services):
~~~
tsp -I ip ... \
    -P fork 'tsp -P zap service1 -O ip ...' \
    -P fork 'tsp -P zap service2 -O ip ...' \
    -P fork 'tsp -P zap service3 -O ip ...' \
    -P zap service4 -O ip ...
~~~

cdavid14, 2019-07-28:
I understand @lelegard , in case I wanted to do a bench test to do what we do today through a closed solution, to situate the context I work on an ISP taking care of the network as a whole, but recently this same ISP bought a TV operation. and I am situated in this new world, I am also an independent developer, my idea is to study this new world and develop solutions, I came across tsduck and I am learning a lot, for this lack of knowledge I have to open this issue to know more as you use it in practice.

Maybe try do something embedded to local market..