tsduck/tsduck #382: Regulation with --pcr-synchronous fails on PCR loopback

mlaurendeau, 2019-07-29:
Hello,

I'm using the following command to stream a low-bitrate ts carrying timed data:

`tsp -v -I file --infinite input.ts -P pcradjust -P continuity --fix -P regulate -p 1 --pcr-synchronous --wait-min 1 -P fork -b 1 "socat -b 188 UDP-SENDTO:235.10.8.3:51083,bind=172.29.42.18 -" -P fork -b 1 "socat -b 188 UDP-SENDTO:235.10.8.4:51084,bind=172.29.42.18 -" -O drop`

The ts contains a single program with a single PID. The data packets are 188-byte length. The ts packet sequence is basically PAT - PMT - PES in a loop.

I observed that regulation fails after about 26.5 hours, which seems to indicate a problem on PCR loopback.

mlaurendeau, 2019-07-29:
I forgot to say I'm using a self-compiled version 3.17-1189.

lelegard, 2019-07-31:
Hi @mlaurendeau 

26.5 hours is exactly one complete PCR round. So, yes this is related.

If fact, the code in class `PCRRegulator` does take care of the PCR value looping back to zero. But there is a bug when the PCR reaches its initial value again, after returning to zero once. This is why the problem appears exactly after 26.5 hours, a complete round, and not when the PCR loops back to zero, somewhere between 0 and 26.5 hours.

Thank you for having detected and reported the problem. I will fix this bug. Sorry for the inconvenience.

lelegard, 2019-08-01:
Hi @mlaurendeau 

The bug has been fixed in commit 9e765f83319738a7aac2641ff932f2e943e27372. I haven't tested it over 26.5 hours however. Could you please confirm if this is correct for you?

lelegard, 2019-08-05:
Version 3.18 has been released. Please open a new issue in case of problem with it.

closed 2019-08-05 by lelegard

mlaurendeau, 2019-08-21:
Hi @lelegard

Sorry for not answering earlier, I had a vacation. I launched a test before my vacation. The issue is no more happening. Thanks for the fix!