tsduck/tsduck #383: dektec DTU-215

khalyngom, 2019-07-29:
Hello
I would like to send data using a DEKTEC DTU-215 via a coaxial cable.
these data will be recovered by a DVB (raspberry)

my problem is: I can not find the code combination to send the data and receive it.
Thank you for your help

rezna11, 2019-07-29:
DTU-215 is a multi standard modulator, so you will need to be more precise in specifying what exactly is it that you can't find. I could always find all necessary parameters in the TSDuck manual.

lelegard, 2019-07-29:
What do you mean by "code combination" ?

You need to use the output plugin `dektec` on the sending system and the input plugin `dvb` on the receiving raspberry (with a DVB-C or DVB-T tuner of course). Then you need to decide which modulation parameters to use (this is your choice, we cannot help). Finally, use the appropriate options on the `dektec` and ` dvb` plugins for these modulation parameters. Use the TSDuck user's guide for these options, there are plenty of them.

khalyngom, 2019-07-29:
@lelegard 

combination of code = instructions for sending datas 

I use this statement to send a text file 
in my network but I have trouble collecting data at my DVB :

  _**tsp -I file  /home/ngom/Desktop/tsp-command.txt -O --device 0 --modulation dvb-T --bandwidth 8 -f 51400000**_

lelegard, 2019-07-29:
The plugin `file` takes a binary transport stream file as input. A file named `tsp-command.txt` is probably not such a file. Please read the TSDuck user's guide.

lars18th, 2019-07-29:
Hi @khalyngom ,

I assume that you like to send UNICAST data from a server with the DVB modulator (DEKTEC DTU-215) and receive it using a DVB tuner (the raspberry with a tuner). That's right?

So, you need this:
1. One tool to send files over unicast UDP.
2. Configure a regular network between both machines (server and receiver).
3. Check that you can send files using the unicast UDP transport.
4. When your network works, then use the TSP tool to encapsulate the unicast UDP stream inside a DVB TS using MPE (read the documentation and examples about the `mpeinject` filter).
5. Use the DEKTEC DTU-215 to send the resulting DVB TS with your encapsulated data.
6. On the other side use the DVB tuner to receive the TS using the TSP tool and decapsulate the MPE unicast (read the documentation and examples about the `mpe` filter).
7. Finally process the unicast UDP stream at destination.

As you can see it's a complex environment and UNIDIRECTIONAL only. However, it's feasible. But if you really need to stablish a network over coaxial I recomment to use some Ether to Coax Adapters, as they can work BIDIRECTIONAL and are more cheap and robust.

Regards.

khalyngom, 2019-08-06:
hello I used this command to inject the flow in my network, now my problem is how to recover the data sent at the raspberry level.

_tsp_ -I file file_name.ts --infinite -O dektec --device 0 --modulation DVB-T --bandwidth 8 -f 514000000_
I managed to stream a feed on my network but I do not know how to get the data from the other side

lelegard, 2019-08-07:
> I managed to stream a feed on my network but I do not know how to get the data from the other side

You need a DVB-T tuner and a receiving software depending on what you want to do. If you want to use TSDuck as a receiver, then use the `dvb` input plugin. At this stage, I can only repeat my advice from a previous post in this thread: read the TSDuck user's guide.