tsduck/tsduck #386: tsp: ip: unknown option --rtp

zozzancs, 2019-07-30:
I'm trying to create an rtp output with this simple command:
tsp -I file tsduck.ts -O ip --rtp 127.0.0.1:10001

I get this error message:
tsp: ip: unknown option --rtp

I found this option on page 171 in the user's guide:
https://tsduck.io/download/docs/tsduck.pdf

Is this an unimplemented feature?
I'm quite new to tsduck, am I missing something?

lelegard, 2019-07-30:
The option `--rtp` is new in version 3.18. The version 3.18 is not yet final but you can build it from the source repo using instructions [here](https://tsduck.io/doxy/building.html).

If you use a binary package for an official release, the documentation which is the package applies to this package. The online version of the user's guide is aligned with the latest source revision.

I plan to release version 3.18 soon, probably within a week or so.

zozzancs, 2019-07-30:
Oh, great, thank you!

closed 2019-07-30 by zozzancs