tsduck/tsduck #387: building last release fails

marcabramson, 2019-07-30:
HI Thierry,
since I need to use the new plugin -psimerge, I have loaded the last source code (and also the last tools), and try to recompile, as indicated there https://tsduck.io/doxy/building.html

unfortunately, doing the build\Build.ps1 on powershell (I'm using windows10) give the following result (after a long compile)

[build_res.txt](https://github.com/tsduck/tsduck/files/3447261/build_res.txt)

is there a way to get a working Tsduck with all the dll including the psimerge option somewhere

thnks in advance for your help

BR
Marc

lelegard, 2019-07-30:
My first question is: did you perform a recent update of Visual Studio ?

I ask this because I ran into the very same problem no later than last night. On a personal Windows 10 VM with Visual Studio 2019, I accepted the latest VS update. Right after this update, I got the exact same error. I think that the updated version was Visual Studio 16.2.0.

On the other hand, on my professional Windows 10 laptop - where I did not update Visual Studio - VS 2019 is still version 16.1.5 (compiler: Visual C++ 2019 00435-60000-00000-AA167), I still continue to compile correctly. And of course, I do not accept any VS update now !

So, it looks like that Visual C++ introduced a bug in its very last version. I have not yet found any time to find to workaround. No need to say that the same code continues to compile correctly with GCC and LLVM (clang) as well as previous versions of VS.

This is not the first time I run into bugs with Visual Studio, one of them being a silent code generation bug which is still not fixed (read comments in file `tsArgMix.h`). Visual C++ is definitely a sh...y compiler...

marcabramson, 2019-07-30:
Re: [tsduck/tsduck] building last release fails (#387)

Bonjour Thierry,

Le mardi 30 juillet 2019 à 16:57:21, vous écriviez :

My first question is: did you perform a recent update of Visual Studio ?

yes, I did JUST  before trying to compile
do you have a way to send me the complete package compiled with your working visual studio, so that I can use the option I need ?

thanks in advance
BR
Marc

here are the info about the Microsoft Visual Studio I now have

Microsoft Visual Studio Community 2019
Version 16.2.0
VisualStudio.16.Release/16.2.0+29123.88
Microsoft .NET Framework
Version 4.7.03056

Version installée : Community

Visual C++ 2019   00435-60000-00000-AA071
Microsoft Visual C++ 2019

Adaptateur de test pour Google Test   1.0
Active l'utilisation des outils de test de Visual Studio avec les tests unitaires écrits pour Google Test. Les conditions d'utilisation et les notifications tierces sont disponibles dans le répertoire d'installation de l'extension.

ASP.NET and Web Tools 2019   16.2.290.48383
ASP.NET and Web Tools 2019

Assistants Microsoft Visual C++   1.0
Assistants Microsoft Visual C++

Extension IntelliCode   1.0
Informations détaillées sur l'extension Visual Studio IntelliCode

Gestionnaire de package NuGet   5.2.0
Gestionnaire de package NuGet dans Visual Studio. Pour plus d'informations sur NuGet, visitez https://docs.nuget.org/

Microsoft JVM Debugger   1.0
Provides support for connecting the Visual Studio debugger to JDWP compatible Java Virtual Machines

Microsoft MI-Based Debugger   1.0
Provides support for connecting Visual Studio to MI compatible debuggers

Outils C#   3.2.0-beta4-19359-03+15b43b33901c88f68ef43f8314b5a2457716780d
Composants C# utilisés dans l'IDE. Selon votre type de projet et vos paramètres, une version différente du compilateur peut être utilisée.

Outils Visual Basic   3.2.0-beta4-19359-03+15b43b33901c88f68ef43f8314b5a2457716780d
Composants Visual Basic utilisés dans l'IDE. Selon votre type de projet et vos paramètres, une version différente du compilateur peut être utilisée.

Package hôte de l'adaptateur de débogage de Visual Studio Code   1.0
Couche d'interopérabilité pour l'hébergement d'adaptateurs de débogage Visual Studio Code dans Visual Studio

Package Microsoft Visual Studio VC   1.0
Package Microsoft Visual Studio VC

ProjectServicesPackage Extension   1.0
ProjectServicesPackage Visual Studio Extension Detailed Info

Test Adapter for Boost.Test   1.0
Enables Visual Studio's testing tools with unit tests written for Boost.Test.  The use terms and Third Party Notices are available in the extension installation directory.

Visual Studio Tools pour CMake   1.0
Visual Studio Tools pour CMake

Visual Studio Tools pour CMake   1.0
Visual Studio Tools pour CMake

I ask this because I ran into the very same problem no later than last night. On a personal Windows 10 VM with Visual Studio 2019, I accepted the latest VS update. Right after this update, I got the exact same error. I think that the updated version was Visual Studio 16.2.0.
On the other hand, on my professional Windows 10 laptop - where I did not update Visual Studio - VS 2019 is still version 16.1.5 (compiler: Visual C++ 2019 00435-60000-00000-AA167), I still continue to compile correctly. And of course, I do not accept any VS update now !
So, it looks like that Visual C++ introduced a bug in its very last version. I have not yet found any time to find to workaround. No need to say that the same code continues to compile correctly with GCC and LLVM (clang) as well as previous versions of VS.
This is not the first time I run into bugs with Visual Studio, one of them being a silent code generation bug which is still not fixed (read comments in file tsArgMix.h). Visual C++ is definitely a sh...y compiler...
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

marcabramson, 2019-07-30:
Re: [tsduck/tsduck] building last release fails (#387)

Bonjour Thierry,
is there an easy way to compile tsduck or windows using GCC ? (to avoid this Visual C++ problem)
BR
Marc

Le mardi 30 juillet 2019 à 16:57:21, vous écriviez :

My first question is: did you perform a recent update of Visual Studio ?
I ask this because I ran into the very same problem no later than last night. On a personal Windows 10 VM with Visual Studio 2019, I accepted the latest VS update. Right after this update, I got the exact same error. I think that the updated version was Visual Studio 16.2.0.
On the other hand, on my professional Windows 10 laptop - where I did not update Visual Studio - VS 2019 is still version 16.1.5 (compiler: Visual C++ 2019 00435-60000-00000-AA167), I still continue to compile correctly. And of course, I do not accept any VS update now !
So, it looks like that Visual C++ introduced a bug in its very last version. I have not yet found any time to find to workaround. No need to say that the same code continues to compile correctly with GCC and LLVM (clang) as well as previous versions of VS.
This is not the first time I run into bugs with Visual Studio, one of them being a silent code generation bug which is still not fixed (read comments in file tsArgMix.h). Visual C++ is definitely a sh...y compiler...
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

lelegard, 2019-07-30:
> is there an easy way to compile tsduck or windows using GCC ? (to avoid this Visual C++ problem)

Do you mean GCC on Windows ? I never tried.

To reduce the risk of inconsistencies and errors, I always use the "natural" compiler on all platforms. The platform / compiler matrix I use is the following:

- Linux + GCC
- macOS + LLVM (clang)
- Windows + Visual C++

There is still a risk for bugs but I think that the number of potential build issues would be higher with "unholy alliances" such as Windows + GCC.

marcabramson, 2019-07-30:
Re: [tsduck/tsduck] building last release fails (#387)

Bonjour Thierry,

Le mardi 30 juillet 2019 à 17:21:03, vous écriviez :

is there an easy way to compile tsduck or windows using GCC ? (to avoid this Visual C++ problem)

Do you mean GCC on Windows ? I never tried.
To reduce the risk of inconsistencies and errors, I always use the "natural" compiler on all platforms. The platform / compiler matrix I use is the following:

Linux + GCC
macOS + LLVM (clang)
Windows + Visual C++

There is still a risk for bugs but I think that the number of potential build issues would be higher with "unholy alliances" such as Windows + GCC.

that's make sense !!!
so, I have to wait until Visual Cstudio is fixed or until you correct the "non" error which is your code::

BR
Marc

—

Cordialement,
 Marc                            e-mail : marc_a@c-kur.tv

lelegard, 2019-07-30:
I will try to find a workaround in the code. There are already a few `#ifdef` in the TSDuck code to deal with various compiler bugs.

I will build binary packages on my not-up-to-date but not-buggy version of Visual Studio and push them tonight.

lelegard, 2019-07-30:
Pre-releases are available here : https://tsduck.io/download/prerelease/

lelegard, 2019-07-30:
And a source code fix has been pushed in the repo.

closed 2019-07-30 by lelegard