tsduck/tsduck #388: mixing 2 ts.. can I control the bitrate ?

marcabramson, 2019-07-31:
Hello Thierry
I have 2 streams
- one called channels.ts which does contains a few clear or ecnrypted channel (and also corresponding PCR)
- a second, update.ts, which does contains my data for updating, using DVB SSU. So, all DSM CC section, plus the corresponding PAT, PMt, and the NIT and BAT with linkage descriptor as specified in ETSI 102 006 chapter 6.1, but NO PCR.

what I need to do is to add my SSU to the channels.. so, I need to merge the stream or to mux them together

I have nearly succeed to do it by using your psimerge command like that
tsp -I file  channels.ts update.ts --label-base 1 -P psimerge --main-label 2 --merge-label 1 -O file result3.ts
(I was surprised that psimerge is merging the psi AND the others PID. I was expecting first to have to do that in 2 steps, first a merge, and then a psimerge, but this is not an issue.)

The problem is that, when I do this command, the packets from the second file are inserted in the resulting file in burst.. for example, my dsmcc section will be in TS packet 21, 23,25, then 33,35,37,39.. So, the interval between the packets is very short,,. and even if the average bitrate is ok, the peak one is to high, which does prevent the section to be correctly loaded... 

What I would need is to control the stream rate in which the packets from the second ts are injected in the first ts..  using either --bitrate  or --inter-packet like it's done in your mux command.. which I can not use directly because some of the PID (like SDT and other) are present in both my stream.

another option could be to use the --interleave option, to define that the 2 ts  have to be interleaved, but to add a value to interleave, such as interleave N (which mean take N packet from 1st TS and then 1 packet from second TS)

do you think you could implement this feature (controlling the bitrate of the "merging") in a near future ? or is there any other way to do what I want to do.?

thanks in advance for your answer.

regards
Marc

lelegard, 2019-07-31:
With the input plugin `file`, the multiple files are sent _in sequence_, complete file 1, then complete file 2, etc. With option `--interleave`, the files are mixed in fixed size chunks. 

To insert an SSU in a stream, I had another approach which is described in section 5.2.15 of the user's guide. Note that this was 10 years ago, so some parts could be probably simplified now.

marcabramson, 2019-07-31:
Re: [tsduck/tsduck] mixing 2 ts.. can I control the bitrate ? (#388)

Bonjour Thierry,
using your method will force me to "split" my ssu ts into multiple bin file
one for the pmt
one for the dsi
one for the dii
one for the ddbs
one for the bat
one for the nit

which will be to difficult to do..

in your mux command, you already have a case uin which the multiple files are not sent in sequnece

what I really need is a mix between mux and psimerge feature.. a psi merge in which we can control the rate..

thanks for your answer anyway.. I will have to find another solution to remux my ssu in the main ts.

BR
Marc

Le mercredi 31 juillet 2019 à 12:16:31, vous écriviez :

With the input plugin file, the multiple files are sent in sequence, complete file 1, then complete file 2, etc. With option --interleave, the files are mixed in fixed size chunks.
To insert an SSU in a stream, I had another approach which is described in section 5.2.15 of the user's guide. Note that this was 10 years ago, so some parts could be probably simplified now.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

marcabramson, 2019-07-31:
Re: [tsduck/tsduck] mixing 2 ts.. can I control the bitrate ? (#388)

Bonjour Thierry,

succeded like that, using 2 tools from my own
one is called removed_pid and replace a pid in a stream by stuffing
one is called extract_pid and extract a pid from a stream

so, I do it like that

rem step 1: extract the pid 101 (dsmcc data) from the update ts
extract_pid update.ts pid101.ts 0101

rem step 2: merge the 2 ts to have a TS which will have the service from both stream in PMT and BAT/not from update stream
tsp -I file --interleave   channels.ts update.ts --label-base 1 -P psimerge --main-label 2 --merge-label 1 -O file result1.ts

rem step 3: remove pid 101 (dsmcc data) from resulting stream, since the bitrate of this pid 101 is not the good one
remove_pid result1.ts result2.ts  101

rem step4: now re-insert pid 101 with the birate I want using mux command and bitrate option
tsp -I file result2.ts  -P mux --bitrate 60000   pid101.ts  -O file result3.ts

Le mercredi 31 juillet 2019 à 12:16:31, vous écriviez :

With the input plugin file, the multiple files are sent in sequence, complete file 1, then complete file 2, etc. With option --interleave, the files are mixed in fixed size chunks.
To insert an SSU in a stream, I had another approach which is described in section 5.2.15 of the user's guide. Note that this was 10 years ago, so some parts could be probably simplified now.
—

Cordialement,
 marc                            e-mail : marc_a@c-kur.tv

marcabramson, 2019-08-01:
Thierry, I have another case in which I need to control the bit rate

I have a first steam, with a bit rate of 65Mb/s, including 40Mb/s of stuffing
I have a second stream, which is (now, thanks to sremove)  a SPTS , with the only existing service with a streamrate of 15,7Mb/s

I want to remux the service from the second TS in  the first TS, but I need to keep the bitrate os the service  so that this service can be correctly played 
I can not use my trick to merge the stream, and then extract video and audio from second stream, and re-insert them using mux command, because this will (probably) destroy the audio/video sync.

Any idea how I can process using tsduck. ?

thanks in advance for your help.