tsduck/tsduck #389: DirectShow BDA Graph - Why use MPEG2 Demux and TIF?

SS-FF, 2019-08-02:
The DirectShow graph is built in a way that the tuner or receiver component is connected (via tee splitter) to both tsDuck's SinkFilter _and_ the MS MPEG2 Demux filter followed by a transport information filter.

At least from my understanding, the MPEG2 Demux filter is used/required to verify that the graph is built correctly and will most likely work.

What I'm wondering though, is whether it is required to keep those filters in the graph while it's running? Wouldn't it be sufficient to use the demux filter for validation only and simply replace it with the SinkFilter afterwards?

The demux and TIF filters are decoding the stream in parallel to tsDuck while the results are never used at all.

Hence my question whether those could be dropped from the graph or if you're keeping them for certain reasons?

lelegard, 2019-08-02:
DirectShow.... What a story. How could I say? I do not pretend to understand it completely. I do not even feel like understanding it. I just hate it!

The version 1 of TSDduck, in 2005, was Linux only. And interfacing tuners on Linux is easy with LinuxTV API: just a couple of ioctl calls. Version 2, in 2008, was meant to be portable and natively compiled on Windows. And I discovered DirectShow... The horror of DirectShow...

Receiving a transport stream from a tuner was never clearly documented as a global use case. So, I had to pick pieces of code and articles here and there to make it work. So, this explains that TSDuck may not use the best method to read the TS from a tuner.

Especially, the most horrifying characteristics of DirectShow was that it was not able to do the simplest thing we can imagine from a tuner: just getting the input stream with zero additional work. So, I was obliged to develop a dedicated DirectShow filter to grab the raw TS data, a so-called "sink filter". And such a simple feature requires thousands of lines of COM programming. I have rarely done such a disgusting work...

I am definitely willing to get good advices from DirectShow experts since I could not get any in the last ten years. Escpecially, getting rid and my custome sink filter would be great.

Now, how did I come to the kind of graph TSDuck currently uses? At the time, I had a few DVB tuners (mainly DVB-S and DVB-T). All of them came with proprietary TV watching applications. So, for each tuner, I used a tool named "DirectShow Spy" while the TV watching application was running. DirectShow Spy displays the graph which is currently used by any running application. And in all cases, they used a Demux filter which remains active all the time.

In TSDuck, unlike TV watching applications, we do not need a demux since we grab the complete stream. But, without a Demux filter, I noticed that I received nothing. But when copied the graph structure of the TV applications, with an otherwise unused Demux filter, it worked.

I never had to idea to remove the Demux filter after starting the capture.

Honestly, I am very cautious and conservative with DirectShow. What works in some config does not with another one. I have learnt it the hard way. I have also noticed that the structure of the graph which was used by some tuners is not the same as other tuners (I do not remember the details, it was ten years ago). So, I had to implement a "try & error" strategy to address the general case: try to build a graph the general way, then try other graph structures if it fails.

I am not really tempted to remove the Demux filter after starting the capture. Even if it works on a couple of tuners I have on a specific Windows version, there is still a risk that it breaks somewhere else for another user.

SS-FF, 2019-08-02:
I feel your pain! It's real and justified.

IMO, DirectShow is a great architecture on its own. But MS had failed so hard to make it accessible and understandable to developers like with no other technology before. (otherwise, MS Dev tools, frameworks and documentation has mostly been superior compared to other platforms - at least from my point of view - at the time when DirectShow was still a top notch technology).

At the same time - 10-15 years ago - I had to develop DirectShow filters for VoIP telephony (audio), and that turned out to be a real nightmare. Not because of DirectShow itself but MS had ended up with 3 or 4 totally different ways for doing DirectShow programming. One was ATL based, another one template-based another one used those base classes - I don't remember all details. 
Whatever you looked up and did research on - it was by guarantee not following the programming model you had chosen.
I switched several times until I was confident about the choice and finally - I got this implemented in a very nice way with little amount of code that's easy to follow. But the amount of time to get there - was a multiple of the amount of time for doing the actual implementation....

Sad story for a great technology IMO.

SS-FF, 2019-08-02:
> I am not really tempted to remove the Demux filter after starting the capture. Even if it works on a couple of tuners I have on a specific Windows version, there is still a risk that it breaks somewhere else for another user.

IIRC it's not possible to change the filter graph while it is running.
What I meant is replacing the demux filter before running the graph but after verifying the connection.

I know DirectShow to some extent but I'm not a high roller there and I'm not an expert for BDA.
Of course I know DirectShowSpy and GraphStudio/Next. Recently I had a look at the filtergraph that the 'dvb viewer' application is creating and it is using a custom sink filter (no mpeg2 demux).
That's why I came to ask about it. Well - and also because of the cpu usage of tsp in this setup.

lelegard, 2019-08-02:
> because of the cpu usage of tsp in this setup.

Do you think that this is due to DirectShow in general and the Demux filter in particular?

Did you do some benchmark?

SS-FF, 2019-08-03:
I haven't tried yet to get rid of the demux filter, so at this time its just speculation.

But I'm sure that it's not DirectShow itself. Even though it appears to be a huge framework and requires a lot of coding - the overhead for running a graph is minimal.

Also I tend to (want to) believe that it's not tsDuck's own code.

But I'll try soon to remove the demux and tif filters and then I'll be able to do some comparison.

lelegard, 2019-08-07:
> Recently I had a look at the filtergraph that the 'dvb viewer' application is creating and it is using a custom sink filter (no mpeg2 demux).

Could you post a screenshot of this graph?

Do you know if their sink filter is doing something special, in addition to passing all media samples to the application?

SS-FF, 2019-08-09:
Meanwhile I found out the relevant facts:

The 'Microsoft Network Provider' requires the TIF filter being present in the graph in order to work.

I had seen two applications running without MP2 demux and TIF filters, but from a second look, it turns out that they are both using custom network provider filters.

So there's no secret at the side of the sink filter - it's the network provider filter. 
What I can't answer yet is the question whether that's the only reason for using a custom Network Provider filter or if there are other benefits in doing so...

Here's one of those graphs:

[image]

oviano, 2019-08-27:
Just a FYI, a while back when I integrated TSDuck into my project, I looked into the whole Windows DShow graph thing quite a lot.

This thread has quite a few graphs from various apps that I grabbed:

https://github.com/tsduck/tsduck/issues/227

And this thread was my (failed!) investigation into some packet loss issues encountered using TSDuck on Windows, seemingly caused by DirectShow.

https://github.com/tsduck/tsduck/issues/235

Just saw this thread and thought it might be of interest, especially the one with all the graphs in it.

SS-FF, 2019-09-05:
@oviano - Thanks for your reply. I had written your conversations before and I need to thank you for your work on supporting ATSC.  I got this working successfully with various HDHR and Hauppauge devices.

lelegard, 2019-10-05:
Hi @SS-FF and @oviano,

You are more experienced than others on DirectShow, so I need your help.

I start to get interested in ISDB. On Linux, a tuner-friendly OS, it is quite easy. On Windows, it looks like DirectShow is now some kind of "abandonware" without support for any recent TV feature and no more programming documentation for the latest version of Windows. Specifically, I see exactly zero documentation on ISDB for DirectShow and the header files only show a few definitions but nothing useful. Additionally, there is no ISDB network provider nor any "ISDB Locator" class. 

But there are supported ISDB tuners, coming with their BDA drivers and DirectShow filters for tuner and network provider.

If we step backward and have a wider look, we do not really need a full DirectShow graph. We just need the tuner filter. But the interface to the tuner filters was never documented. The only documented interface was the network providers and "network locators". This is what TSDuck is using, requiring a complete graph.

So, on one hand, most network provider filters are standard Microsoft filters and, one the other hand, the tuner filters are always proprietary. So, there must be some standard interface between the network provider filters and the tuner filters. And all tuner vendors know this interface because they can write tuner filters which correctly inter-operate with standard network provider filters from Microsoft.

But where is the definition of this interface between network provider filters and tuner filters? 

It must be somewhere, but where?

If tuner vendors can have it, why cant't we have it?

Thanks for you feedback.

oviano, 2019-10-06:
Well anything I learnt about DirectShow / BDA I learned from looking at your code so I'm not sure if I can be much help.

I have been searching for the last hour on Google, to see if I can find any useful references, and you are right, it's a bit of a black hole out there.

Maybe the samples referenced in this link might contain something useful though?

https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/sample-bda-drivers

SS-FF, 2019-10-14:
@lelegard 

> But where is the definition of this interface between network provider filters and tuner filters?
> 
> It must be somewhere, but where?
> 
> If tuner vendors can have it, why cant't we have it?

Of course you can have it. It's in the Windows DDK (Device Driver Development Kit) which is the starting point for any driver development.

The relevant DDK versions for BDA driver information are:
- 7600.16385.1 
(IIRC it's the last version including the BDA driver samples
- 3790.1830
Older, but might still be of interest as it has a completely different set of sample

Trying to access that filter directly in some way is not a path worth following. It will fail in most cases.
There are different architectures, some BDA drivers are single-filter while others are using a filter-pair (tuner + capture filter). Others are even modeling CI modules as an additional filter.
For the data flow between those filters, you can't even know whether it happens in user-mode or in kernel-mode or just in hardware (a pin-connection doesn't mean that the data needs to flow through software). Of course at some point, the data needs to travel from kernel mode to user mode. This is done under the hood by the kernel streaming proxy.

If you would really want to do something totally custom, you would rather create your own custom filters everywhere around the bda driver filters.

But I don't think that you even have to, as I got a certain theory...

SS-FF, 2019-10-14:
### THEORY: ISDB-T and ISDB-S are regularly supported on Windows via BDA

I'm not able to proof it as I don't have any ISDB device, but I'll tell how I come to this:
- MS has developed a full set of service information parsers for ISDB. They wouldn't have developed and published that, when ISDB wasn't officially supported. (https://docs.microsoft.com/en-us/previous-versions/windows/desktop/api/dvbsiparser/nn-dvbsiparser-iisdb_bit)
- Hardware vendors are offering ISDB hardware with BDA drivers
- In the forums of TV software applications you can find a number of stories where it was stated that they don't support ISDB and users later reported that they got their ISDB hardware to work.
In two cases, the developers were surprised and said that it would only work because the driver would 'mimic' a DVB-T device for compatibility.
I believe that those assumptions are wrong and that ISDB devices are working just regularly
- Look at the following registry key: Computer\HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Multimedia\TV\Tuning Spaces
- Open GraphEdit or GraphStudioNext
  - Add MS Network Provider
  - Add DVB-S or DVB-T BDA tuner filter
  - Open the property pages for the network provider
  - In the list of tuning spaces, you will see ISDB-T or -S in addition to DVB-T or -S
- I don't have as much knowledge about DVB as you, Thierry, but from what I've read is that the physical part of ISDB is almost identical to DVB, so I suppose it just wasn't necessary to develop separate locator and tuning objects for ISDB

So, that's the indications that made me come to that theory.

If I'm right, you would simply use the corresponding DVB-x interfaces for tuning spaces and locator objects...

SS-FF, 2021-05-03:
I guess this is all done already..

closed 2021-05-03 by SS-FF