tsduck/tsduck #392: Building 3.18-1361 with Visual Studio 2017, but projects all require tools v142...?

oviano, 2019-08-13:
Not sure if I'm overlooking something, but I've got to downgrade all the projects to tools v141 in order to build on VS2017, correct?

I'm not quite ready to upgrade to VS2019 yet.

oviano, 2019-08-13:
Also, after changing all projects to v141, I get build errors (or maybe one repeated), which I have posted here:

https://github.com/tsduck/tsduck/issues/313

oviano, 2019-08-14:
Solution seems to be just a case of disabling warning C4623 across some/all projects.

But out-of-the-box the latest source does not build on VS2017/WIn10 without the above two changes.

lelegard, 2019-08-14:
Hi @oviano 

Thank you for reporting this. Maintaining a generic configuration for Visual Studio is a challenge I never really mastered, despite many efforts to manually structure `.vcxproj`  and  `.props` files.

I no longer have VS 2017 and, since I use the Community edition, I do not know how to explicitly install VS 2017 side by side with VS 2019. Let's try to find a common unique configuration which equally works with my VS 2019 and your VS 2017. Any VS 2015 around ?

I have made some changes in commit 168160c8354a12ca0165edd8f52dc4f6382ae8b5 which could fix the issue. It definitely works on VS 2019. Please test on VS 2017, without modification.

There was an attempt to compute the right PlatformToolset in a configuration file but it apparently failed to address all cases. And v142 was only a fallback solution, when everything else failed. Now, I have used another method and removed the hard-coded fallback.

I have also disabled the warning from the VS header files which appeared in VS 2017 but not in VS 2019.

oviano, 2019-08-14:
Thanks - I manually applied those changes to the 3.18-1361 code and I can confirm that it now correctly builds with no warnings or errors on VS2017/WIn10.

lelegard, 2019-08-14:
Thanks for your feedback.

closed 2019-08-14 by lelegard