tsduck/tsduck #393: Clarification of changes to API now that ts::DuckContext exists?

oviano, 2019-08-14:
So my old pseudo-code is:

- create report
- create report handler (and make report use it)
- create COM
- create and open tuner
- tune, capture, demux, etc....
- cleanup tuner, etc
- delete COM
- delete report, report handler

So this becomes:

- create report
- create report handler (and make report use it)
- create COM
- create DuckContext (and pass in report)
- create and open tuner
- tune, capture, demux, etc....(passing in DuckContext where it doesn't compile otherwise)
- cleanup tuner, etc
- delete DuckContext
- delete COM
- delete report, report handler

I.e. the DuckContext is just an internal thing for the library, that it uses to maintain state for various things, and I don't need to concern myself with anything more than creating one, and passing it around where required?

I noticed it mentions in the documentation "- Default region for UHF and VHF frequency layout." as one of the things it stores.

I should still be using ts::HFBand::SetDefaultRegion() though, correct?

lelegard, 2019-08-14:
> I.e. the DuckContext is just an internal thing for the library, that it uses to maintain state for various things, and I don't need to concern myself with anything more than creating one, and passing it around where required?

Yes. You may ignore it after creation. But you can also get benefit from it if you like.

Too many different specific contexts had to be maintained. This is why I introduced `DuckContext`, a global context class which maintains all contextual information a typical TV application needs to maintain. And which can receive more features in the future without breaking the API again.

It contains initial information which can be defined and parsed from the command. To make it easy for applications, you can define several set of command-line options using `defineOptionsForDVBCharset(Args&)`, `defineOptionsForPDS(Args&)`, `defineOptionsForStandards(Args&)` or `defineOptionsForHFBand(Args&)`. Thus, each application can define consistent sets of options, depending on what it works on. Then, after command parsing, the applications can load options values and automatically initialize the context using `loadOptions(Args&)`.

The `DuckContext` object is also used to enrich the context and provide more pertinent information later. For instance, if a demux encounters a DVB-specific or ATSC-specific table, it adds the information that it is in DVB and/or ATSC context. When a non-MPEG descriptor is encountered in some MPEG table later (typically a PMT), the deserialization of the PMT knows if the descriptor should be interpreted as a DVB or ATSC one. ISDB or other standards could be added later.

> I should still be using ts::HFBand::SetDefaultRegion() though, correct?

You can still use the static methods in `HFBand`. But they are global to the application. If this is fine for you application, you can proceeed as previously.

But it is now recommended to use `setDefaultHFRegion()` from a `DuckContext` object and then use `defaultHFRegion()`, `vhfBand()` and `uhfBand()` from the same context object. For instance, each `tsp` plugin has its own context. So, distinct plugins in the same instance of `tsp` can define the same sets of options such as default HF-band options and use their own distinct defaults. Using `ts::HFBand::SetDefaultRegion()` would affect the whole application, meaning all plugins running in the same instance of `tsp`.

oviano, 2019-08-14:
Ok, thank you.

So I'm setting the default region because (if I recall from a few months ago) it was needed internally somewhere to map frequencing/channels, so if I don't set this in the DuckContext, then the code will pick up the static one I assume? Or the DuckContext will default its own one to the static one, and then the mapping will use the DuckContext one....well, ok I can look that up myself I guess :o)

closed 2026-04-24 by github-actions[bot]