tsduck/tsduck #394: Ts file to Dektec modulator

kapook007, 2019-08-16:
i have some problem about  tsduck setting

TSDuck - The MPEG Transport Stream Toolkit - version 3.18-1361
Dektec dta-111

mytsfile.TS with Dektec streamxpress to dvb-t is ok

ts file to multicast is ok
tsp -v -I file --infinite mytsfile.TS -P regulate -O ip 224.1.1.111:5555

ts file to Tsduck Dektec modulator not working i got error
tsp -v -m -I file mytsfile.TS -P regulate -O dektec --modulation DVB-T --frequency 626000000
* dektec: Got CPU underflow.
* dektec: Got FIFO underflow.
* dektec: Got CPU underflow.
* dektec: Got FIFO underflow.

i know about bitrate
mytsfile bitrate 13,723,040 bps
dektec: initial output bitrate: 27,144,385 b/s
i try to fix with
tsp -v -m --bitrate 27144385 -I file mytsfile.TS -P regulate -O dektec --modulation DVB-T --frequency 626000000

this step no error but in tv reciver video play not smoth play fast and lag and some time sound stop !!
how to fix this problem

this is my log

tsp -v -m --bitrate 27144385 -I file mytsfile.TS -P regulate -O dektec --modulation DVB-T --frequency 626000000
* regulate: minimum packet burst duration is 2,000,000 nano-seconds
* file: initial input bitrate is 27,144,385 b/s
* dektec: using Dektec device 0, output channel 0 (DTA-111 port 1)
* dektec: using DVB-T FEC 3/4, bandwidth 8-MHz, constellation 64-QAM, guard 1/32, transmission 8K
* dektec: setting output TS bitrate to 27,144,385 b/s
* dektec: setting output carrier frequency to 626,000,000 Hz
* dektec: output fifo size: 16,777,216 bytes, max: 16,777,216 bytes, typical: 16,777,216 bytes
* dektec: initial output bitrate: 27,144,385 b/s
* dektec: Will preload FIFO before starting transmission. Preload FIFO size: 13,421,772 bytes.
* regulate: regulated at bitrate 27,144,385 b/s

lelegard, 2019-08-16:
Hi @kapook007,

You should remove the regulate plugin when using a modulator as output.

You need it when using the ip output because the ip output is not self-regulated, the UDP packets are sent immediately. Without regulate, the packets would be read from the input file and immediately sent over IP at the speed of the input disk or the output network, whichever is slower. So, the output speed would be way too fast. The plugin regulate is here to slow down the transmission of packets at the playout speed, based on the estimated bitrate.

On the other hand, a Dektec modulator is self-regulated. Packets are output exactly at the speed which is computed from the modulation parameters. Input packets from the disk are waiting for the modulator. Using the plugin regulate is at best useless (the regulation is done in the modulator) and at worse counter-productive. In fact, the precision of the regulate plugin is based on an estimation of the bitrate which is given from the input plugin. Here, the estimation is based on PCR's. The precision is not perfect. The plugin may sometimes wait too much or not enough, creating small bursts and small underflows. The average is correct and you don't see the difference with the ip output. On the other hand, the Dektec modulator is extremely precise and reports the smallest underflows, which in fact is what you see.

So, just remove the regulate plugin from the command with the dektec output.

kapook007, 2019-08-17:
i try to remove  -P regulate  but same problem

2 file use with Dektec streamxpress ok

file1 - video lag
https://youtu.be/1i2BSuNodJk
file2 - video fast , lag
https://youtu.be/IbQeqx7pd_U

tsp -v -m --bitrate 27144385 -I file mytsfile.TS -O dektec --modulation DVB-T --frequency 626000000

* file: initial input bitrate is 27,144,385 b/s
* dektec: using Dektec device 0, output channel 0 (DTA-111 port 1)
* dektec: using DVB-T FEC 3/4, bandwidth 8-MHz, constellation 64-QAM, guard 1/32, transmission 8K
* dektec: setting output TS bitrate to 27,144,385 b/s
* dektec: setting output carrier frequency to 626,000,000 Hz
* dektec: output fifo size: 16,777,216 bytes, max: 16,777,216 bytes, typical: 16,777,216 bytes
* dektec: initial output bitrate: 27,144,385 b/s
* dektec: Will preload FIFO before starting transmission. Preload FIFO size: 13,421,772 bytes.
* [MON] 2019/08/17 09:24, resource monitoring started
* dektec: DTA-111 output FIFO load is 13,421,696 bytes, starting transmission
* [MON] 2019/08/17 09:24, VM:42 MB (+4,096 B), CPU:26.79% (average:26.79%)
* [MON] 2019/08/17 09:24, VM:42 MB (leaking), CPU:26.56% (average:26.67%)
* [MON] 2019/08/17 09:24, VM:42 MB (leaking), CPU:21.68% (average:25.01%)
* [MON] 2019/08/17 09:24, VM:42 MB (leaking), CPU:22.15% (average:24.29%)
* [MON] 2019/08/17 09:24, VM:42 MB (leaking), CPU:25.43% (average:24.52%)
* [MON] 2019/08/17 09:25, VM:42 MB (-49,152 B), CPU:20.62% (average:23.87%)
* [MON] 2019/08/17 09:25, VM:42 MB (leaking), CPU:21.10% (average:23.47%)
* [MON] 2019/08/17 09:25, VM:42 MB (leaking), CPU:22.81% (average:23.39%)
* [MON] 2019/08/17 09:25, VM:42 MB (leaking), CPU:22.45% (average:23.29%)
* [MON] 2019/08/17 09:25, VM:42 MB (leaking), CPU:21.84% (average:23.14%)
* [MON] 2019/08/17 09:25, VM:42 MB (leaking), CPU:21.68% (average:23.01%)
* [MON] 2019/08/17 09:26, VM:42 MB (leaking), CPU:19.82% (average:22.74%)
* [MON] 2019/08/17 09:27, VM:42 MB (leaking), CPU:24.19% (average:23.22%)
* [MON] 2019/08/17 09:28, VM:42 MB (stable), CPU:22.68% (average:23.09%)
* [MON] 2019/08/17 09:29, VM:42 MB (+32 kB), CPU:21.11% (average:22.69%)
* [MON] 2019/08/17 09:30, VM:42 MB (+28 kB), CPU:22.91% (average:22.73%)

lelegard, 2019-08-17:
Hi @kapook007,

OK, I did not realise from your original post that your input file had an incorrect bitrate.

First, without the plugin regulate, you apparently no longer have underflow message. This is what I explained. Good.

Second, the transmission is ok but the video is incorrect. You do not specify any modulation parameter. The defaults are displayed here:
~~~
* dektec: using DVB-T FEC 3/4, bandwidth 8-MHz, constellation 64-QAM, guard 1/32, transmission 8K
~~~
The bitrate from those defaults is in fact 27144385 b/s:
~~~
$ tsterinfo -w 8 -h 3/4 -g 1/32 -c 64
Transport stream bitrate: 27,144,385 b/s
~~~

Your input file must be CBR and must  _exactly_ match the expected bitrate. Since they do not, from the bitrates you mention, then the modulator sends a file at a bitrate X while the file should be played at 27144385 b/s and the video cannot be smooth.

To output a TS on a modulator with TSDuck, you need a clean CBR stream at the exact bitrate. There are two ways to get such a stream:

1. Use a MUX to build this stream. Note that TSDuck is a not a MUX, so you need another tool.
2. Capture a real-world DVB-T stream with the same modulation parameters and transform it while preserving the bitrate (no packet should be removed, they should be turned into stuffing). This is what I typically do.

If you cannot get clean input files with the right bitrate, you can do the following:

- Remove the `--bitrate` option from `tsp` because is indicates an incorrect value for the input stream. Let `tsp` figure out the bitrate by itself, using PCR's.
- Use again the `regulate` plugin. This time, it will be used to regulate at the layout speed _of the input stream_, not the output speed, which is incorrect for you stream.
- Use option `--stuffing` with the output plugin dektec. Thus, the modulator will add stuffing instead of reporting underflow.

This will not be perfect. The video will be at correct speed but the transmission will contain bursts. You may smoothen the transmission a bit using smaller bursts in the regulate plugin (option `-p`). But if you need better, then get a CBR input stream at the correct bitrate.

kapook007, 2019-08-19:
i try to this command not working
tsp -v -m -I file mytsfile.TS -P regulate -O dektec --stuffing --modulation DVB-T --frequency 626000000

i try to use 3rd mux ( 3 udp spts out 1 tpts fix 27,144,385 b/s )
i use 3rd mpts_mux_winapp http://www.scientamedia.com/mpts_mux_winapp.html
this command
tsp -v -m -I ip  239.1.1.1.111:1234 -O dektec --modulation DVB-T --frequency 626000000

now tsduck is  working 
mpts_mux_winapp  is demo lic and very small sofware 1MB
and githun no opensource  mux with bitrate controle !!!!!

thanks for your hellp