tsduck/tsduck #395: DTA 115 UHF

michaeladams123, 2019-08-18:
Hi,
Kindly how much of channels hd with bandwidth 2.5 mega for each channel can i used for uhf and vhf with card dectek 115 rev3

and as below the sample command for one channel but how for multiple channels same time ?

tsp -v -I dektec --device 0 --modulation DVB-T --frequency 506000000 --dvbt-bandwidth 8

lelegard, 2019-08-18:
The capabilities of the Dektec modulator is a question for Dektec. 

The [description the DTA-115](https://www.dektec.com/products/PCI/DTA-115/) seems to indicate only one output. So, you can produce only one channel at a time.

The [DTA-2115B](https://www.dektec.com/products/PCIe/DTA-2115B/) can produce up to 8 distinct channels at the same time, provided that you purchase the appropriate license (the default version is single channel only).

The TSDuck output plugin `dektec` uses one output, meaning one channel. You can produce several outputs using several concurrent `tsp` commands. But if you use a DTA-2115B, I do not know how it is identified, one device or eight devices, I have no idea.

michaeladams123, 2019-08-18:
thanks for your kindly support,

https://github.com/tsduck/tsduck/wiki/Muxing-use-case:-create-live-streams-from-files

I found this  sample for old version DTA 111 and used two channels same this possible and correct on DTA 115 ?
and if we can use input multiple  udp or http channels

lelegard, 2019-08-18:
You seem to use the words "channel" and "bandwidth" in a confusing way. You must be more specific.

- In the UHF and VHF bands, a _channel_ is a frequency (along with a _bandwidth_). So, using two channels means two TS on two distinct frequencies, two modulators (maybe on the same physical output as with the DTA-2115B).
- In the Wiki article you mention, "channel" is used in the common sense of "TV channel", a word we do not use in engineering. MPEG uses "program" and DVB uses "service". To dimension a service, use the word "bitrate", not bandwidth. The Wiki article describes a way to mux two services into one single TS to be sent to one single modulator.

Since you mentioned the VHF band, I interpreted it the first way.

Saentist, 2019-08-26:
> thanks for your kindly support,
> 
> https://github.com/tsduck/tsduck/wiki/Muxing-use-case:-create-live-streams-from-files
> 
> I found this sample for old version DTA 111 and used two channels same this possible and correct on DTA 115 ?
> and if we can use input multiple udp or http channels

DTA-111 can modulate single Transport stream, I can't believe it care what information is in stream.
aka 
Multiplex all channels in one transport stream then send stream to modulator

to be simple to be understand
Modulator need prepared Transport Stream with specific bandwidth (transfered packets per second)
modulator itself send this muxed data to RF (coaxial cable) with fixed parameters with need to be equal to bandwidth of TS data (there is Bitrate Calculators)
read this
https://github.com/tsduck/tsduck/wiki/Dektec-use-case:-Multicast-(CBR)-to-DVB-T

using this 
https://tsduck.io/streams/
in my case i use France: Terrestrial network snapshot

```tsp -v -I file fr-tnt.ts -i -O dektec --device 0 --modulation DVB-T --frequency 177500000 --bandwidth 7 -g 1/32 -r 7/8```

you can tune this to your tv directly after modulator