tsduck/tsduck #397: Can build current source

joolzg, 2019-08-26:
[LD] release-x86_64/tsanalyze
  [LD] release-x86_64/tsbitrate
  [LD] release-x86_64/tscmp
  [LD] release-x86_64/tsdate
  [LD] release-x86_64/tsdektec
  [CXX] tsdump.cpp
tsdump.cpp: In constructor ‘Options::Options(int, char**)’:
tsdump.cpp:66:1: error: ‘Options::duck’ should be initialized in the member initialization list [-Werror=effc++]
 Options::Options(int argc, char *argv[]) :

gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609

lelegard, 2019-08-27:
The code was just modified and compiled on Mac with clang / LLVM last night. I usually fix this type of compiler issue on GCC and MSVC afterwards. Will do...

joolzg, 2019-08-27:
Tried with clang as well, same outcome.

lelegard, 2019-08-27:
> Tried with clang as well, same outcome.

I confirm that it compiles without error on macOS 10.14.6 with clang 10.0.1.

Anyway, the error is justified. I added the missing initialiser and pushed the code. It now compiles on Ubuntu 19.04 with GCC 8.3.0.

closed 2019-08-27 by lelegard